coolsnowwolf/lede #3374: SSR Plus +添加Trojan节点无法播放youtube的视频

nathanxie, 2020-02-27:
昨天软路由更新固件，用SSR Plus +添加Trojan节点遇到以下问题，请大佬麻烦帮我看下。
1、连接Trojan节点后，谷歌搜索无法打开。
2、油管首页能正常打开，并能正常搜索视频内容，但是点击播放跳转到播放页面就一直无法正常播放，我查看视频速度（详细统计信息），显示没有速度。
3、我通过SSR Plus +设置全局模式进行网页测速和fast测试的时候，均能跑满我的300m宽带，测试ip也都是我vps的ip。
4、我用了3个种模式，gfw和绕过大陆，还有全局模式。只有全局模式能正常测速，以上1，2点还是存在。

SharerMax, 2020-02-27:
#3359

coolsnowwolf, 2020-02-27:
卸载 smartdns 之类就好了

nathanxie, 2020-02-27:
> 卸载smartdns之类就好了

好的，非常感谢，今天下班回去卸载smartdns之后看是否能解决问题。

nathanxie, 2020-02-27:
@coolsnowwolf 你好，我刚到家测试，按照你的方法删除smartdns之后，重启openwrt，问题还是一样，没有得到解决。

9l, 2020-02-27:
你设置的什么dns

yifang5775, 2020-02-27:
我也遇到类似问题，我用的是JMS的SS机场，经常不能正常打开youtube,有时又能打开，而谷歌等其他都能正常打开。

nathanxie, 2020-02-27:
> 你设置的什么dns

[image]

就是是用默认的，另外没有做其他dns设置

hcym, 2020-02-27:
jms家的只有一个GIA啦，以前是三个，😭

sharkskin, 2020-02-28:
看看是不是ipv6的原因，卡youtube我遇到过，惯了ipv6，重新连接wifi就好了

kevinlo0909, 2020-02-28:
经本人实验楼主可以试试我的方法，开启tls和填写证书，然后在windows下ipconfig /flushdns清空dns缓存，这样设置以后就可以正常了

kevinlo0909, 2020-02-28:
我未清空dns之前问题和楼主的一模一样，清空客户端的dns之后就正常了希望能解决楼主的问题

lloyd18, 2020-02-28:
遇到同样问题

hcym, 2020-02-28:
路由器上没有问题，没用TLS，就是域名密码😁

nathanxie, 2020-02-28:
> 经本人实验楼主可以尝试我的方法，开启tls和填写证书，然后在windows下ipconfig / flushdns清空dns缓存，这样设置以后就可以正常了

谢谢你的帮助，应该是这个问题，作为小白来说，如何开启TLS和填写证书呢？我用的一键安装，电脑用客户端使用没有这个问题哦。

guochan2019, 2020-02-28:
我的问题跟楼主也差不多,更新固件,VPS也重新部署了,但切换trojan后，在LUCI测试不通，但网页可以打开油管，就是无法播放

kevinlo0909, 2020-02-28:
> > 经本人实验楼主可以尝试我的方法，开启tls和填写证书，然后在windows下ipconfig / flushdns清空dns缓存，这样设置以后就可以正常了
> 
> 谢谢你的帮助，应该是这个问题，作为小白来说，如何开启TLS和填写证书呢？我用的一键安装，电脑用客户端使用没有这个问题哦。

这个问题来源于dns问题 经过再次测试 不开tls 和 不填写证书路径 都可以正常链接，就是国内淘宝直通车链接打不开，暂时确定还是dns问题

kevinlo0909, 2020-02-28:
> 我的问题跟楼主也差不多,更新固件,VPS也重新部署了,但切换trojan后，在LUCI页面测试连接错误，但网页可以打开油管，就是无法播放

在windows下ipconfig / flushdns清空dns缓存，这样设置以后就可以正常了

guochan2019, 2020-02-28:
> > 我的问题跟楼主也差不多,更新固件,VPS也重新部署了,但切换trojan后，在LUCI页面测试连接错误，但网页可以打开油管，就是无法播放
> 
> 在windows下ipconfig / flushdns清空dns缓存，这样设置以后就可以正常了
谢谢兄弟回复，都试过了，LuCI界面就显示不通，我专门刷新旧固件来比较的，显然是源码哪儿出了问题，希望能修复吧

SharerMax, 2020-02-28:
@guochan2019 按你的描述还是dns的问题，尝试在PC端和路由器中用dig活或nslookup查看www.google.com的IP地址是否一致

guochan2019, 2020-02-28:
> 昨天软路由更新固件，用SSR Plus +添加Trojan节点遇到以下问题，请大佬麻烦帮我看下。
> 1、连接Trojan节点后，谷歌搜索无法打开
> 2、油管首页能正常打开，并能正常搜索视频内容，但是点击播放跳转到播放页面就一直无法正常播放，我查看视频速度（详细统计信息），显示没有速度
> 3、我通过SSR Plus +设置全局模式进行网页测速和fast测试的时候，均能跑满我的300m宽带，测试ip也都是我vps的ip
> 4、我用了3个种模式，gfw和绕过大陆，还有全局模式。只有全局模式能正常测速，以上1，2点还是存在

今晚重新编译，同样的配置，居然好了，供参考

nathanxie, 2020-02-29:
[image]
[image]
[image]
自己又折腾了下，问题还没有解决，请问是我域名解析的问题吗？

guochan2019, 2020-02-29:
你在DHCP/DNS的高级设置里把禁止解析 IPv6 DNS 记录这个选中，看有没变化

keeno1982, 2020-03-06:
遇到同样的问题，默认配置，无其它DNS设置，SSR模式正常，Trojan连通性检查失败，无法访问谷歌，可以打开油管但是无法播放。已经编译最新版本，还是不行。OpenWrt R20.3.5 / LuCI Master (git-20.062.50551-a6189c5)

closed 2020-03-06 by coolsnowwolf

gause555, 2020-03-12:
我也有这个问题，新刷固件，SMARTDNS没启用，直接配置了trojan就出现youtube能正常打开，就是点开浏览视频一直是黑屏。

thoryyll, 2020-07-20:
同病相怜，我用openclash和passwal也是出现同样的问题。。

hcym, 2020-07-20:
完全没有问题，clash和pass也不错啊😂

wsl610036, 2020-07-22:
> > 昨天软路由更新固件，用SSR Plus +添加Trojan节点遇到以下问题，请大佬麻烦帮我看下。
> > 1、连接Trojan节点后，谷歌搜索无法打开
> > 2、油管首页能正常打开，并能正常搜索视频内容，但是点击播放跳转到播放页面就一直无法正常播放，我查看视频速度（详细统计信息），显示没有速度
> > 3、我通过SSR Plus +设置全局模式进行网页测速和fast测试的时候，均能跑满我的300m宽带，测试ip也都是我vps的ip
> > 4、我用了3个种模式，gfw和绕过大陆，还有全局模式。只有全局模式能正常测速，以上1，2点还是存在
> 
> 今晚重新编译，同样的配置，居然好了，供参考

遇到了同样的问题，不知道怎么解决，其他外网都正常就谷歌一套不行

myi101, 2020-08-02:
我也是同样的问题，我的是N1软路由，固件是F大的40+，Trojan链接在SSR plus里面完美，在Passwall里面可以打开Youtube，但是网页视频就是黑屏的，统计信息里面网速很低，另两项是没有的

myi101, 2020-08-02:
你好，我在SSR Plus里面是完美使用的，在Passwall里面可以打开Youtube，但是网页视频就是黑屏的，统计信息里面网速很低，另两项是没有的。现在找到了解决办法，Passwall模式的DNS解析模式是ChinaDNS-NG，我改成了和SSR PLUS一样的PDNSD了就正常了。

hlz781105, 2020-09-10:
我也遇到这样的问题了，解决的办法是在光猫后面加了一个路由器拨号就好了，我的路由器是华硕，没有改固件，用的是官方原版固件。
    LEDE开始是安装在一个树莓派4B，以为是4B的问题，然后找了一个闲置的主机，安装了LEDE，还是一样。只好安装QT5使用。后来增加了几个无线设备，感觉光猫的无线有些带不动了，打10000号把光猫拨号改桥接，然后接了一个路由器，想了一下，把树莓派也接上了。然后设置好路由器PPPOE，上网，网关改树莓派IP居然一切正常了。撤了路由器，找10000号改光猫拨号，还是照旧，不能用。再打10000号，改桥接，接上路由器，正常了。

aonima, 2020-11-16:
房东换了条网线和路由器，也出现相同问题了，请教下有没有解决方案？谢谢啦

myi101, 2020-11-17:
我现在改用trojan-go了， passwall 新版才支持。换插件就好了 在2020年11月16日 16:56，aonima 写道： 

房东换了条网线和路由器，也出现相同问题了，请教下有没有解决方案？谢谢啦

[

{

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/coolsnowwolf/lede/issues/3374#issuecomment-727835541",

"url": "https://github.com/coolsnowwolf/lede/issues/3374#issuecomment-727835541",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

laughman56, 2021-07-23:
不知道楼主还在吗？我刚刚也是遇到这个问题了，我也是软路由openwrt下SSR，windows不能上谷歌和能打开YouTube网页但看不了视频，但安卓端没有问题，ipconfis/flushdns命令也打过了，最后解决问题的是吧ipv6协议给取消了

Cosmos-Break, 2022-04-03:
是ipv6的问题，会解析到youtube的ipv6地址，然后就打不开视频了。说到底还是dns的问题。

n1k0ver3E, 2022-04-07:
same problem，使用openwrt，在网络的dhcp/dns中 - 高级设置 - 选中 禁止解析 IPv6 DNS 记录，并重启旁路由
即可解决