coolsnowwolf/lede #3379: 这是什么原因的错误？

lizhenglin2002, 2020-02-27:
Makefile:307: recipe for target '/home/esir/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built' failed
make[3]: *** [/home/esir/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built] Error 1
make[3]: Leaving directory '/home/esir/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#2.05#0.59#17.30
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/esir/lede'
package/Makefile:107: recipe for target '/home/esir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/esir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/esir/lede'
/home/esir/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

jzh33, 2020-02-27:
我也遇到过这个问题，但我不是编译x86_64，最后取消upx压缩后不报错了

lizhenglin2002, 2020-02-27:
> 我也遇到过这个问题，但我不是编译x86_64，最后取消upx压缩后不报错了

请问在哪里取消

jzh33, 2020-02-27:
貌似是在network=>project v=v2ray,我也记不太清楚了

lizhenglin2002, 2020-02-28:
> 貌似是在network=>project v=v2ray,我也记不太清楚了

closed 2020-02-28 by lizhenglin2002