coolsnowwolf/lede #3380: Unable to locate package linux-libc-dev:i386

lujimmy, 2020-02-27:
Ubuntu 18.04.4 LTS根据 README执行
_sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib linux-libc-dev:i386_
时报“Unable to locate package linux-libc-dev:i386”错误

boatmac, 2020-02-27:
试试：

sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade

EnnawYang, 2020-02-27:
ubuntu18不需要linux-libc-dev:i386

lujimmy, 2020-02-27:
> 试试：
> 
> sudo dpkg --add-architecture i386
> sudo apt-get update
> sudo apt-get upgrade
> sudo apt-get dist-upgrade

已解决。但具体有无影响不得而知，之前在google cloud上编译时没注意到该错误，也编译成功了。今天折腾用github actions来编译，直接报错中止了。

closed 2020-02-27 by lujimmy