coolsnowwolf/lede #3381: 请问大佬 为啥我编译官方版openwrt 时n2n和ramfree 是英文界面

vbzc, 2020-02-27:
大佬 你好，我把n2n和ramfree 放入到官方的package目录下，编译后发现 这两个软件的界面是英文的，以前我记得是中文的，而且 opkg list-installed 命令，没有找到luci-i18n-n2n_v2-zh-cn和 luci-i18n-ramfree-zh-cn这两个文件。

请问大佬 是不是 我哪里没做对，导致没有中文呀？谢谢

hue715, 2020-02-29:
官方默认的luci.mk是检索po文件夹下的zh_Hans，大佬的po文件夹下中文为zh-cn，需要修改。

vbzc, 2020-03-01:
@hue715 谢谢答复，我改了文件夹的名称，重新编译后发现 还是英文。

hue715, 2020-03-01:
> @hue715 谢谢答复，我改了文件夹的名称，重新编译后发现 还是英文。

我试了一下，把文件夹zh-cn改成zh-Hans，配置文件中的中文可以选上。（luci > modules > translations > chinese（zh_Hans））。你看看自己的配置文件中有没有。
[image]
[image]

vbzc, 2020-03-02:
@hue715 谢谢回复，原来是我把安装码包放在openwrt/package/zc 目录下，造成编译后没有中文界面的。移到 openwrt/package下，就可以了。以前我也是这样做的，没问题呀，不知道怎么回事？

closed 2020-03-06 by coolsnowwolf