coolsnowwolf/lede #3382: 编译出错

xiexianghua, 2020-02-27:
In file included from ../deps/v8/src/base/bits.h:9:0,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/c++/7/type_traits:38:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
tools/v8_gypfiles/v8_libbase.host.mk:152: recipe for target '/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o' failed
make[5]: *** [/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0'
Makefile:150: recipe for target '/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/.built' failed
make[3]: *** [/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/xiexianghua95/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.98#0.17#4.50
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/xiexianghua95/lede'
package/Makefile:107: recipe for target '/home/xiexianghua95/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/xiexianghua95/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xiexianghua95/lede'
/home/xiexianghua95/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

brick713, 2020-02-27:
+1 同样的问题

lqzhj, 2020-02-27:
2. 命令行输入 sudo apt-get update ，然后输入
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib linux-libc-dev:i386

试试

asushugo, 2020-02-27:
看看是不是gcc-multilib或者g++-multilib没安装
sudo apt-get install gcc-multilib g++-multilib

xiexianghua, 2020-02-28:
可以了

closed 2020-02-28 by xiexianghua