coolsnowwolf/lede #3383: github Actions 编译，x64正常，d2编译报错！要怎么调整.yml?

d975xbx, 2020-02-27:
[image]
[image]

shake, 2020-02-27:
需要设置cpu。这是我极路由3的文件，参考一下。
https://github.com/shake/lede/blob/master/.github/workflows/openwrt-ci.yml

d975xbx, 2020-02-27:
感谢回复，我的.config文件的cpu架构是对的，本地选择后上传的
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="ramips"
CONFIG_TARGET_SUBTARGET="mt7621"
CONFIG_TARGET_PROFILE="DEVICE_d-team_newifi-d2"
CONFIG_TARGET_ARCH_PACKAGES="mipsel_24kc"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc"
CONFIG_CPU_TYPE="24kc"
CONFIG_LINUX_4_14=y

closed 2020-02-28 by d975xbx