coolsnowwolf/lede #3385: 最新965924a，卡wget编译不过

Jennal, 2020-02-27:
```
make[3]: Entering directory '/root/lede/feeds/packages/net/wget'
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/.pkgdir/wget/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.wget_installed
mkdir -p /root/lede/bin/targets/x86/64/packages /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget/CONTROL /root/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget/usr/bin
install -m0755 /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/src/wget /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget/usr/bin/wget-ssl
find /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package wget is missing dependencies for the following libraries:
libcrypto.so.1.0.0
libssl.so.1.0.0
Makefile:109: recipe for target '/root/lede/bin/packages/x86_64/packages/wget_1.20.3-2_x86_64.ipk' failed
make[3]: *** [/root/lede/bin/packages/x86_64/packages/wget_1.20.3-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/root/lede/feeds/packages/net/wget'
time: package/feeds/packages/wget/ssl/compile#0.65#0.48#1.02
package/Makefile:111: recipe for target 'package/feeds/packages/wget/compile' failed
make[2]: *** [package/feeds/packages/wget/compile] Error 2
make[2]: Leaving directory '/root/lede'
package/Makefile:107: recipe for target '/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
/root/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

ckxt, 2020-02-28:
把 .config文件删掉，然后make clean后make menuconfig再试试

Jennal, 2020-02-29:
> 把 .config文件删掉，然后make clean后make menuconfig再试试

make clean以后可以了，非常感谢

closed 2020-02-29 by Jennal