coolsnowwolf/lede #3387: 今天更新，重新编译，出现错误：/usr/include/c++/7/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory

AmibaWorld, 2020-02-27:
make[3]: Leaving directory '/home/v2ray/lede/feeds/luci/applications/luci-app-upnp'
time: package/feeds/luci/luci-app-upnp/compile#1.74#1.44#3.88
In file included from ../deps/v8/src/base/bits.h:9:0,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/c++/7/type_traits:38:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/7/stdlib.h:36:0,
                 from ../deps/v8/include/v8-platform.h:10,
                 from ../deps/v8/src/base/bounded-page-allocator.h:8,
                 from ../deps/v8/src/base/bounded-page-allocator.cc:5:
/usr/include/c++/7/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.

AmibaWorld, 2020-02-27:
thanks
https://github.com/coolsnowwolf/lede/issues/3382#issuecomment-591984394

看看是不是gcc-multilib或者g++-multilib没安装
sudo apt-get install gcc-multilib g++-multilib
解决了

closed 2020-02-27 by AmibaWorld

n2ray, 2020-02-28:
回报一下，按以下处理后成功编译
[#3382 (comment)](https://github.com/coolsnowwolf/lede/issues/3382#issuecomment-591984394)