coolsnowwolf/lede #3393: 编译报错

brick713, 2020-02-28:
( cd /home/XXX/lede/build_dir/hostpkg/go-1.13.6/src ; GOROOT_FINAL=/home/XXX/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/XXX/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/XXX/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap.
Building Go toolchain1 using /home/XXX/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap.
# bootstrap/cmd/compile/internal/ssa
/home/XXX/lede/build_dir/hostpkg/go-1.13.6/src/cmd/compile/internal/ssa/rewritegeneric.go:2907[/home/XXX/lede/build_dir/hostpkg/go-1.13.6/pkg/bootstrap/src/bootstrap/cmd/compile/internal/ssa/rewritegeneric.go:2911]: internal compiler error: out of memory
go tool dist: FAILED: /home/XXX/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 2
Makefile:271: recipe for target '/home/XXX/lede/build_dir/hostpkg/go-1.13.6/.built' failed
make[3]: *** [/home/XXX/lede/build_dir/hostpkg/go-1.13.6/.built] Error 2
make[3]: Leaving directory '/home/XXX/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#74.10#11.42#87.78
package/Makefile:111: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/XXX/lede'
package/Makefile:107: recipe for target '/home/XXX/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/XXX/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/XXX/lede'
/home/XXX/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

之前编译一直没有问题，突然报这个错误

coolsnowwolf, 2020-02-28:
out of memory 内存不足

brick713, 2020-02-28:
> out of memory 内存不足

设备没有变化是 1G 的内存，之前编译一直正常。

brick713, 2020-02-28:
更换成4G 内存正常了。

closed 2020-02-28 by brick713