coolsnowwolf/lede #3399: 新编译UPNP无法使用。

BlackTouma, 2020-02-28:
**平台：MTK**
**设备：斐讯K2P**

**问题：**昨天新编译的固件，UPNP开启，但无法生效使用，列表一直为空。

SharerMax, 2020-02-28:
[image]

hhyygg2009, 2020-02-28:
我也是，以前那个版本是支持bitcomet的

hyird, 2020-02-28:
> 我也是，以前那个版本是支持bitcomet的

我今天试了下，现在那个也是可以的

naoki66, 2020-02-28:
新版本无问题，也解决了MWAN兼容性（需要自己更新mwan），挺好的。

chenhao6938, 2020-02-28:
我也发现了，upnp不起作用了，迅雷一只没有映射端口。

LeeMenHin, 2020-02-29:
有什么软件可以测试

loonyd, 2020-02-29:
> 有什么软件可以测试

我是用群晖的外部访问那里的路由器设置来测试的，如果upnp ok的话会直接找到upnp路由器

fonlan, 2020-02-29:
> > 有什么软件可以测试
> 
> 我是用群晖的外部访问那里的路由器设置来测试的，如果upnp ok的话会直接找到upnp路由器

那个好像是群晖不支持Openwrt的upnp，换成爱快群晖就能正常识别了。Openwrt的话你试下爱奇艺，应该会创建名为HCDN的upnp服务。

loonyd, 2020-02-29:
> > > 有什么软件可以测试
> > 
> > 
> > 我是用群晖的外部访问那里的路由器设置来测试的，如果upnp ok的话会直接找到upnp路由器
> 
> 那个好像是群晖不支持Openwrt的upnp，换成爱快群晖就能正常识别了。Openwrt的话你试下爱奇艺，应该会创建名为HCDN的upnp服务。

支持的，我用koolshare的去年9.30的2.33版完美支持，我另外一台k2p用的l大的20.2.15编译的固件也是完美支持，但有的就不支持，我自己也不会编译，所以只能一个一个尝试

loonyd, 2020-02-29:
> > > > 有什么软件可以测试
> > > 
> > > 
> > > 我是用群晖的外部访问那里的路由器设置来测试的，如果upnp ok的话会直接找到upnp路由器
> > 
> > 
> > 那个好像是群晖不支持Openwrt的upnp，换成爱快群晖就能正常识别了。Openwrt的话你试下爱奇艺，应该会创建名为HCDN的upnp服务。
> 
> 支持的，我用koolshare的去年9.30的2.33版完美支持，我另外一台k2p用的l大的20.2.15编译的固件也是完美支持，但有的就不支持，我自己也不会编译，所以只能一个一个尝试

[image]

DavidDen, 2020-03-01:
> 新版本无问题，也解决了MWAN兼容性（需要自己更新mwan），挺好的。

想问问是在/feeds/packages/net/mwan3/makefile里面作出更改嘛？因为里面没有网址，我看不到最新的版本号。请问能否提供一下版本号？谢谢

closed 2020-03-05 by BlackTouma