coolsnowwolf/lede #3400: SSR Plus无法使用sni proxy解锁的Netflix节点。

fefz, 2020-02-28:
**_前置条件：_**
有两台server，A和B，A的ip无法正常观看Netflix的节目，B可以正常观看，于是使用了https://github.com/ab77/netflix-proxy 这个项目，可以通过A设置Netflix的DNS指向B的方式来观看Netflix。

**_遇到的问题：_**
在A上面已经搭建好Trojan/V2ray，本地使用SSR Plus的时候，打开Netflix还是提示**error:m7111-5059**，无法观看，但是在局域网内iphone设备使用**Quantumult/Surge**时，同样使用的A节点，Netflix所有节目都可以观看（包括自制剧和非自制剧），证明解锁服务是正常的。

**_尝试解决：_**
起初以为是SSR Plus默认规则的问题，于是把iPhone端的规则在SSR Plus设置了一遍，但还是无法正常观看。

添加的黑名单域名如下：
```
netflixdnstest9.com
netflixdnstest8.com
netflixdnstest7.com
netflixdnstest6.com
netflixdnstest5.com
netflixdnstest4.com
netflixdnstest3.com
netflixdnstest2.com
netflixdnstest1.com
netflixdnstest0.com
netflix.com
netflix.net
nflxext.com
nflximg.com
nflximg.net
nflxso.net
nflxvideo.net
```

添加的强制走代理的IP：
```
23.246.0.0/18
37.77.184.0/21
45.57.0.0/17
64.120.128.0/17
66.197.128.0/17
108.175.32.0/20
192.173.64.0/18
198.38.96.0/19
198.45.48.0/20
```

abclai, 2020-02-28:
复制下 某文章的观点，看有大佬能解决不！
如何正确配置客户端与服务端
对于远程服务器，应该配置 DNS 劫持（即将 53 端口所有流量转发给 DNS 解锁的 DNS 上），可以使用 iptables 和 DNAT 实现。

对于代理客户端，首先应该推荐使用 Clash 和 Surge，正如我前文所说，这两个代理客户端均支持反查 TCP 连接的域名、并将域名发送给代理服务端。如果当前没有在使用 Clash 或 Surge 的用户遇到无法通过 DNS 解锁访问流媒体服务的，应该尝试使用 Clash 和 Surge、排查问题是否出在代理客户端上。
对于 不确定是否支持反解、发送域名给代理服务端 的代理客户端，如路由器上的 KoolSS、FancySS 等，DNS 模式应该选择 dns2socks 或 ss-tunnel（如果是 Padavan，应勾选启用 dnsproxy），因为这些模式都会将对国外域名的 DNS 查询转发到远程服务器，而此时如果远端服务器上正确配置了 DNS 劫持，DNS 解锁的 DNS 也可以收到查询请求，最终也可以在远端服务器上实现流媒体分流。在这些客户端上不应该使用 chinadns 等直连进行 DNS 查询的模式（chinadns 不支持通过代理进行 DNS 查询，因此解析请求不会经过远端服务器）和 PCap_DNSProxy 等 DNS 加密模式（远端服务器无法识别加密后的 DNS 解析请求、无法转发给 DNS 解锁）。我和几个商业性质的公共代理服务提供商交流的结果是，Lean 的 luci-app-ssr-plus 几乎一定会导致 DNS 解锁失效。

本文采用 CC BY-NC-SA 4.0 许可协议，著作权归作者所有。商业转载请联系作者获得授权，非商业转载请注明出处。
作者：Sukka
来源：简析代理客户端导致的远端服务器流媒体 DNS 解锁失效的可能原因 | Sukka's Blog
链接：https://blog.skk.moe/post/how-proxy-client-cause-netflix-unlock-to-fail/

SharerMax, 2020-02-28:
luci-app-ssr-plus中pdnsd目前支持自定义dns，方便dns解锁流媒体。
如果可能的话可以尝试一下自定义dns

fefz, 2020-02-28:
> luci-app-ssr-plus中pdnsd目前支持自定义dns，方便dns解锁流媒体。
> 如果可能的话可以尝试一下自定义dns

是全部域名转发到自定义DNS地址吗？

fefz, 2020-02-28:
> 复制下 某文章的观点，看有大佬能解决不！
> 如何正确配置客户端与服务端
> 对于远程服务器，应该配置 DNS 劫持（即将 53 端口所有流量转发给 DNS 解锁的 DNS 上），可以使用 iptables 和 DNAT 实现。
>
> 对于代理客户端，首先应该推荐使用 Clash 和 Surge，正如我前文所说，这两个代理客户端均支持反查 TCP 连接的域名、并将域名发送给代理服务端。如果当前没有在使用 [...]

Clash我试了，是可以正常使用解锁的，准备在openwrt试试openClash

SharerMax, 2020-02-28:
> > luci-app-ssr-plus中pdnsd目前支持自定义dns，方便dns解锁流媒体。
> > 如果可能的话可以尝试一下自定义dns
> 
> 是全部域名转发到自定义DNS地址

> > luci-app-ssr-plus中pdnsd目前支持自定义dns，方便dns解锁流媒体。
> > 如果可能的话可以尝试一下自定义dns
> 
> 是全部域名转发到自定义DNS地址吗？

和预设的dns一样，只有gfw列表和强制走代理的域名会通过代理转发到指定的dns

fefz, 2020-02-28:
> > > luci-app-ssr-plus中pdnsd目前支持自定义dns，方便dns解锁流媒体。
> > > 如果可能的话可以尝试一下自定义dns
> > 
> > 
> > 是全部域名转发到自定义DNS地址
> 
> > > luci-app-ssr-plus中pdnsd目前支持自定义dns，方便dns解锁流媒体。
> > > 如果可能的话可以尝试一下自定义dns
> > 
> > 
> > 是全部域名转发到自定义DNS地址吗？
> 
> 和预设的dns一样，只有gfw列表和强制走代理的域名会通过代理转发到指定的dns

试过了，还是不行。现在确认的是openclash的fake-ip模式是可以的，但是理论上passwall的dns2socks应该也可以，但是测试结果也是不行。

fefz, 2020-03-01:
才看到要编译srelay，ssr plus才可以支持socks代理，使用dns2socks做pdnsd的上游解决问题了。

closed 2020-03-01 by fefz

tefiszx, 2021-05-20:
关注此问题，我也是最近用DNS+SNI Proxy的方式解锁了netflix才遇到此问题的。目前在vps上设置v2ray的内建dns，强制分流netflix的域名解析。所以用ssr-plus的v2ray节点可以解锁奈飞，而用navie的节点则不能解锁。

xxiaocheng, 2021-05-24:
同样的问题。同样的节点，直接移动端或者pc就可以解锁Netflix，但是路由器上就不行

tefiszx, 2021-05-24:
> 关注此问题，我也是最近用DNS+SNI Proxy的方式解锁了netflix才遇到此问题的。目前在vps上设置v2ray的内建dns，强制分流netflix的域名解析。所以用ssr-plus的v2ray节点可以解锁奈飞，而用navie的节点则不能解锁。

在节点设置里，把PDNSD的域名自定义为解锁IP问题解决。不过不能针对特定域名列表进行DNS分流，还是存在安全隐患。问题是解决了，但是我还是不打算这么用。继续用v2的节点，用v2分流dns吧。