coolsnowwolf/lede #3404: 大佬我找到了wrt1900系列的新驱动，就是不会添加进去

segadrive, 2020-02-28:
https://github.com/eduperez/mwlwifi_LEDE/releases/tag/a2fd00b，请问如何添加，旧驱动有断流问题，感激不尽了

segadrive, 2020-02-28:
https://davidc502sis.dynamic-dns.net/releases/#1900acv2

nivalxer, 2020-02-28:
官方源也更新了：https://github.com/kaloz/mwlwifi
看了一下日志，主要是更新的88W8964固件，也就是WRT3200ACM和WRT32X
更新之后短暂试了一下，还是没解决2.4G的WMM跟IOT设备无线的兼容性问题（无法获取DHCP地址，关闭WMM才行）

segadrive, 2020-02-28:
> 官方源也更新了：https://github.com/kaloz/mwlwifi
> 看了一下日志，主要是更新的88W8964固件，也就是WRT3200ACM和WRT32X
> 更新之后短暂试了一下，还是没解决2.4G的WMM跟IOT设备无线的兼容性问题（无法获取DHCP地址，关闭WMM才行）

我一楼的网址有网友编译的驱动，ipk也安装进去了，结果还是不行

nivalxer, 2020-02-28:
> > 官方源也更新了：https://github.com/kaloz/mwlwifi
> > 看了一下日志，主要是更新的88W8964固件，也就是WRT3200ACM和WRT32X
> > 更新之后短暂试了一下，还是没解决2.4G的WMM跟IOT设备无线的兼容性问题（无法获取DHCP地址，关闭WMM才行）
> 
> 我一楼的网址有网友编译的驱动，ipk也安装进去了，结果还是不行

两边更新日志一样的，没看到有啥特别的改动
我在用wrt32x，也在关注这个问题。我这边主要还有物联网设备连接问题。

segadrive, 2020-02-28:
> > > 官方源也更新了：https://github.com/kaloz/mwlwifi
> > > 看了一下日志，主要是更新的88W8964固件，也就是WRT3200ACM和WRT32X
> > > 更新之后短暂试了一下，还是没解决2.4G的WMM跟IOT设备无线的兼容性问题（无法获取DHCP地址，关闭WMM才行）
> > 
> > 
> > 我一楼的网址有网友编译的驱动，ipk也安装进去了，结果还是不行
> 
> 两边更新日志一样的，没看到有啥特别的改动
> 我在用wrt32x，也在关注这个问题。我这边主要还有物联网设备连接问题。
https://github.com/eduperez/mwlwifi_LEDE/releases 这个网址你看了没有大佬？

segadrive, 2020-02-28:
> > > 官方源也更新了：https://github.com/kaloz/mwlwifi
> > > 看了一下日志，主要是更新的88W8964固件，也就是WRT3200ACM和WRT32X
> > > 更新之后短暂试了一下，还是没解决2.4G的WMM跟IOT设备无线的兼容性问题（无法获取DHCP地址，关闭WMM才行）
> > 
> > 
> > 我一楼的网址有网友编译的驱动，ipk也安装进去了，结果还是不行
> 
> 两边更新日志一样的，没看到有啥特别的改动
> 我在用wrt32x，也在关注这个问题。我这边主要还有物联网设备连接问题。

https://davidc502sis.dynamic-dns.net/releases/这个网址你看了没有大佬，这里面我刷的固件就没毛病，延迟很低，无断流，信号杠杠的

nivalxer, 2020-02-28:
> https://davidc502sis.dynamic-dns.net/releases

谢谢，空了我试试这个固件看看能不能解决我的问题。我之前找了很久解决方案，后面都放弃了。

segadrive, 2020-02-28:
> > https://davidc502sis.dynamic-dns.net/releases
> 
> 谢谢，空了我试试这个固件看看能不能解决我的问题。我之前找了很久解决方案，后面都放弃了。

解决了给我说下啊，帮我迁一下驱动到lean的固件里面来，我实在不会

nivalxer, 2020-02-28:
> > > https://davidc502sis.dynamic-dns.net/releases
> > 
> > 
> > 谢谢，空了我试试这个固件看看能不能解决我的问题。我之前找了很久解决方案，后面都放弃了。
> 
> 解决了给我说下啊，帮我迁一下驱动到lean的固件里面来，我实在不会

mwlwifi两边源码一致，mwlwifi_lede只是预编译好的。
然后
https://davidc502sis.dynamic-dns.net/releases就是openwrt官方源码编译的。
wrt1900断流的问题，我问一下有设备的朋友，使用的openwrt官方固件的确没有太大的断流问题。
wrt32x我试了一下官方固件，物联网设备连接的问题还存在，我就刷回来了，断流问题可能时间比较短，没有遇到。

segadrive, 2020-02-28:
> mwlwifi两边源码一致，mwlwifi_lede只是预编译好的。
> 然后 [...]

有没有方法可以用openwrt的源码把lean的包编译进去的教程，我编译openwrt试试

osxdiy, 2020-02-29:
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
exit 0

segadrive, 2020-02-29:
> echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
> exit 0
thank you,谢谢你啊

segadrive, 2020-02-29:
> 有没有方法可以用openwrt的源码把lean的包编译进去的教程，我编译openwrt试试

大佬，原因找到了，说是要禁用amsdu，你会不会操作？

segadrive, 2020-02-29:
> # 将您的自定义命令放在这里，应该执行一次
> # 系统初始化完成。默认情况下，此文件不执行任何操作。
> 回声0> / sys /内核/调试/ ieee80211 / phy0 / mwlwifi / tx_amsdu
> 回声0> / sys /内核/调试/ ieee80211 / phy1 / mwlwifi / tx_amsdu
> 退出0

我还是不会操作，迷糊了/

nivalxer, 2020-02-29:
翻了一下，看到是有类似提到过这个问题：https://github.com/kaloz/mwlwifi/issues/207
先SSH上去执行
echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
然后看看效果
可以的话放到 系统-启动项，底部的本地启动脚本里面

segadrive, 2020-02-29:
> 翻了一下，看到是有类似提到过这个问题：[kaloz/mwlwifi#207](https://github.com/kaloz/mwlwifi/issues/207)
> 先SSH上去执行
> echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
> 然后看看效果
> 可以的话放到 系统-启动项，底部的本地启动脚本里面

好的👌

segadrive, 2020-02-29:
> > 翻了一下，看到是有类似提到过这个问题：[kaloz/mwlwifi#207](https://github.com/kaloz/mwlwifi/issues/207)
> > 先SSH上去执行
> > echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> > echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
> > 然后看看效果
> > 可以的话放到 系统-启动项，底部的本地启动脚本里面
> 
> 好的

不行，lede根目录根本就没有这些东西

nivalxer, 2020-02-29:
> > > 翻了一下，看到是有类似提到过这个问题：[kaloz/mwlwifi#207](https://github.com/kaloz/mwlwifi/issues/207)
> > > 先SSH上去执行
> > > echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> > > echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
> > > 然后看看效果
> > > 可以的话放到 系统-启动项，底部的本地启动脚本里面
> > 
> > 
> > 好的
> 
> 不行，lede根目录根本就没有这些东西

可以啊
[image]

segadrive, 2020-03-01:
> > > > 翻了一下，看到是有类似提到过这个问题：[kaloz/mwlwifi#207](https://github.com/kaloz/mwlwifi/issues/207)
> > > > 先SSH上去执行
> > > > echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> > > > echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
> > > > 然后看看效果
> > > > 可以的话放到 系统-启动项，底部的本地启动脚本里面
> > > 
> > > 
> > > 好的
> > 
> > 
> > 不行，lede根目录根本就没有这些东西
> 
> 可以啊
> [image]

我再试试

segadrive, 2020-03-01:
> > > > > 翻了一下，看到是有类似提到过这个问题：[kaloz/mwlwifi#207](https://github.com/kaloz/mwlwifi/issues/207)
> > > > > 先SSH上去执行
> > > > > echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> > > > > echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
> > > > > 然后看看效果
> > > > > 可以的话放到 系统-启动项，底部的本地启动脚本里面
> > > > 
> > > > 
> > > > 好的
> > > 
> > > 
> > > 不行，lede根目录根本就没有这些东西
> > 
> > 
> > 可以啊
> > [image]
> 
> 我再试试

关闭了，但是不行，https://davidc502sis.dynamic-dns.net/releases/#1900acv2 他的固件就没问题，暂时用他的把

closed 2020-03-06 by coolsnowwolf