coolsnowwolf/lede #3405: 请问大佬为什么最后个支持环境装不上？

zenghuangjie, 2020-02-28:
xxx@xxx:~$ sudo apt-get -y install linux-libc-dev:i386
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-libc-dev:i386

yyleis, 2020-02-29:
冒号改成点:
sudo apt-get -y install linux-libc-dev.i386

zenghuangjie, 2020-02-29:
> 
> 
> 冒号改成点:
> sudo apt-get -y install linux-libc-dev.i386

emmm，看起来不是这样，我昨天去ubuntu的论坛里问了，答案是这样的：
sudo dpkg --add-architecture i386
sudo apt update
sudo apt install linux-libc-dev:i386

yyleis, 2020-02-29:
你看我的：
yylei@WORK-PC:~$ sudo apt-get install linux-libc-dev:i386
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-libc-dev:i386
yylei@WORK-PC:~$ sudo apt-get install linux-libc-dev.i386
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'linux-libc-dev-i386-cross' for regex 'linux-libc-dev.i386'
Note, selecting 'linux-libc-dev-i386-dcv1' for regex 'linux-libc-dev.i386'
Note, selecting 'linux-libc-dev-i386-cross' instead of 'linux-libc-dev-i386-dcv1'
linux-libc-dev-i386-cross is already the newest version (4.15.0-35.38cross1.1).
0 upgraded, 0 newly installed, 0 to remove and 128 not upgraded.

EnnawYang, 2020-02-29:
> 你看我的：
>
> ```
> yylei@WORK-PC:~$ sudo apt-get install linux-libc-dev:i386
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> E: Unable to locate package [...]

别瞎指挥了好吧

zenghuangjie, 2020-02-29:
> 别瞎指挥了好吧

看你的就看你的，你要说瞎指挥我就很不爱听。
什么叫瞎指挥。冷雪狼的代码明明是:i386，你非要改成.i386。我按ubuntu论坛的代码执行，明明可以装上:i386，我干嘛要去装.i386？

closed 2020-02-29 by zenghuangjie

EnnawYang, 2020-02-29:
> 看你的就看你的，你要说瞎指挥我就很不爱听。
> 什么叫瞎指挥。冷雪狼的代码明明是:i386，你非要改成.i386。我按ubuntu论坛的代码执行，明明可以装上:i386，我干嘛要去装.i386？

亲您看我针对的是您吗？

zenghuangjie, 2020-02-29:
> 亲您看我针对的是您吗？

QAQ!!
真是抱歉，很不好意思，没有仔细看你的引用对象。