coolsnowwolf/lede #3416: 我发的新3有线连接不停断开的问题为啥被删了

chenhao6938, 2020-02-29:
搜了一下，这个问题从17年就有了，op更新了这么久还有这问题？
newifi3刷入op，有线连接不停地断开然后又连上。
系统日志
mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
反复这两条。

hcym, 2020-02-29:
这种软故障根本就没必要问，原因千奇百怪，别人也没法回答，很多是兼容性问题。我是硬件型号多，直接替换排查，换固件最简单。

Aes64X, 2020-02-29:
我的Newifi 2 没有发现这个问题

chenhao6938, 2020-02-29:
> 这种软故障根本就没必要问，原因千奇百怪，别人也没法回答，很多是兼容性问题。我是硬件型号多，直接替换排查，换固件最简单。

刷过无数个op都这样，而且网上也有其他人有这毛病，怀疑是op驱动兼容性问题。但是基于op的潘多拉就没问题，很神奇。

hcym, 2020-02-29:
那不就结了，我的新三也有，不光新三，一堆硬路由和n1蜗牛也有，但是电脑关机重启就好了，热重启都未必行，不是一台哦，四五台，网线也换过，不光断线，网口有时100有时协商1000，现在手头有重要的事时就干脆用无线。

hgh188, 2020-02-29:
5635.769971] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[ 5639.085528] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[ 6784.735200] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[ 6793.920624] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[ 6798.559399] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[ 6801.300612] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[ 7365.508814] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[10936.279894] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[10941.467666] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[10944.333355] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[10948.858884] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[10951.505926] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[16534.698958] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[16538.026671] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[17783.103467] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[62784.449660] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[62789.489605] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[62792.327573] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[62796.900750] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[62799.511815] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[65830.828686] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[65836.387050] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[65839.738080] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[65877.167275] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up

一样的问题 开始我以为是我超频的原因

ntlf9t, 2020-02-29:
我觉得是新3网口供电不足

hgh188, 2020-02-29:
[80666.077989] usb 1-1: USB disconnect, device number 3
[80666.174803] [EXFAT] trying to unmount...
[80666.178894] [EXFAT] unmounted successfully
[80706.732861] ksmbd: kill_server_store:480: kill command received
我拨了移动硬盘 看看有没有疗效  应该是这个原因

chenhao6938, 2020-02-29:
> 那不就结了，我的新三也有，不光新三，一堆硬路由和n1蜗牛也有，但是电脑关机重启就好了，热重启都未必行，不是一台哦，四五台，网线也换过，不光断线，网口有时100有时协商1000，现在手头有重要的事时就干脆用无线。
都快成玄学了。op咋会这样，以前一直用潘多拉，看op可定制性强打算折腾一下呢，结果还是不行。暂时用老毛子算了，还算稳定。

chenhao6938, 2020-02-29:
> [80666.077989] usb 1-1: USB disconnect, device number 3
> [80666.174803] [EXFAT] trying to unmount...
> [80666.178894] [EXFAT] unmounted successfully
> [80706.732861] ksmbd: kill_server_store:480: kill command received
> 我拨了移动硬盘 看看有没有疗效 应该是这个原因

usb我什么都没挂，而且插件基本也没加，就一个科学上网。

XeonLang, 2020-02-29:
你这可能真不是固件的问题，我的新三官方op从18到19一路没有一个固件有这问题

chenhao6938, 2020-02-29:
> 你这可能真不是固件的问题，我的新三官方op从18到19一路没有一个固件有这问题

你试试第三方op看，我其他固件也没问题的。话说你的官方op在哪儿下的？一直还没用过官方的呢。

hcym, 2020-02-29:
其实这种事我也烦躁过，可能和网卡有关，因为替换各种组合，不光mtk系列x86和arm还有高通路由器，只有pc网卡都是螃蟹的，用ar系列和ax200无线网卡一点毛病没有，干脆没有无线的电脑都配ax200，几个六十块钱，提前走进新时代😂

chenhao6938, 2020-02-29:
> 其实这种事我也烦躁过，可能和网卡有关，因为替换各种组合，不光mtk系列x86和arm还有高通路由器，只有pc网卡都是螃蟹的，用ar系列和ax200无线网卡一点毛病没有，干脆没有无线的电脑都配ax200，几个六十块钱，提前走进新时代

第一次遇到路由器挑网卡的。。。应该是固件挑网卡。

eekon, 2020-02-29:
把你手里的新三扔掉吧

DongfeiSay, 2020-02-29:
我这个目前没有这个问题

XeonLang, 2020-03-03:
> > 你这可能真不是固件的问题，我的新三官方op从18到19一路没有一个固件有这问题
> 
> 你试试第三方op看，我其他固件也没问题的。话说你的官方op在哪儿下的？一直还没用过官方的呢。

openwrt.org就能下，新3一直都有稳定版，需要的插件用官方的SDK和L大的源码编译就好了

AmadeusGhost, 2020-03-03:
mtk_soc_eth 这个问题目前4.14内核无解

closed 2020-03-06 by coolsnowwolf