coolsnowwolf/lede #3422: 通过解锁网易云音乐实现回家科学上网

Marcotso, 2020-02-29:
环境描述：固件版本OpenWrt R20.2.15 / LuCI Master (git-20.058.13156-2faf063)
UnblockNeteaseMusic-0.24.1-1
luci-app-ssr-plus 154-1

家中OP 的ssr-plus插件正常启动，设置国内白名单，家中可以正常科学上网；
UnblockNeteaseMusic插件正常启动，不修改设置，默认端口：5200；
安卓手机在外连接WIFI,把WIFI进阶选项中手动设置代理服务器，PROXY主机填写家中OP的DDNS域名，端口填写5200。设置完毕后，安卓手机即可通过回家实现科学上网。

请教大雕，这情况是否安全？毕竟不需要密码

adore595, 2020-02-29:
你们啊 总是能玩出新花样

lizhenglin2002, 2020-02-29:
这操作不错，反正有个ip就行

WROIATE, 2020-02-29:
为啥要这样，直接用路由器带的本地ssr不就好了吗，网易云这个是http代理ssr是sock5代理他不香吗

nnssz, 2020-02-29:
一举两得😊
就是不知道有没有副作用

9l, 2020-03-01:
那这白嫖服务估计活不久

R3pl4c3r, 2020-03-01:
> 环境描述：固件版本OpenWrt R20.2.15 / LuCI Master (git-20.058.13156-2faf063)
> UnblockNeteaseMusic-0.24.1-1
> luci-app-ssr-plus 154-1
> 
> 家中OP 的ssr-plus插件正常启动，设置国内白名单，家中可以正常科学上网；
> UnblockNeteaseMusic插件正常启动，不修改设置，默认端口：5200；
> 安卓手机在外连接WIFI,把WIFI进阶选项中手动设置代理服务器，PROXY主机填写家中OP的DDNS域名，端口填写5200。设置完毕后，安卓手机即可通过回家实现科学上网。
> 
> 请教大雕，这算另一种科学上网姿势吗？这情况是否安全？毕竟不需要密码

直接用ssr plus自带的ssr服务器和socks服务器它不香吗？

StoneForests, 2020-03-01:
你这样很不安全啊，等于是把socks5端口暴露在外，而且没有任何认证。。不建议把网易云的这个端口暴露在公网上，如果你想远程回家，可以用ssrp+插件中的ssr服务端，或者v2ray server开一个服务端，再用代理软件将网易云app流量强制代理回家不是更好么。。

honjow, 2020-03-01:
你这不就是开了个端口在公网。。。而且还是没啥保护的

KFERMercer, 2020-03-01:
上街不穿护屌布

closed 2020-03-06 by coolsnowwolf

hjh142857, 2020-05-03:
> 直接用ssr plus自带的ssr服务器和socks服务器它不香吗？

> 你这样很不安全啊，等于是把socks5端口暴露在外，而且没有任何认证。。不建议把网易云的这个端口暴露在公网上，如果你想远程回家，可以用ssrp+插件中的ssr服务端，或者v2ray server开一个服务端，再用代理软件将网易云app流量强制代理回家不是更好么。。

所以请教下各位大佬。。用ssr服务端连回家了但网易云不ok了，应该用什么姿势正确解锁呢

lizhenglin2002, 2020-05-03:
可以试试Socks5服务端模式，可以设置账户名和密码