coolsnowwolf/lede #3423: 大家好，请问一下如何让固件适合在国外使用呢？

liruishengus, 2020-02-29:
最近我朋友想在HK家里弄一个软路由，之后让国内的朋友也可以通过他的软路由来看youtube和netflix。已采购硬件3215U+4网口，全新安装20200226版本并放在香港家里那边，使用的网络是PCCW；
但是最近我下载论坛里面很多大神编译的固件安装后，发现以下几个问题：
1：无法通过openwrt访问s.1688.com网站，ping不通这个域名，导致我朋友想回国做商务采购做不到，但换回最原始的TPLINK就可以直接访问到了，我没有开启任何插件的，连防火墙都关闭了。
2：看到很多插件，只看到SSR+里面有个服务器端设置，但是里面的加密/协议/混淆都只有一种两种，并没有那些机场常用的加密/协议/混淆的选项选择，是不是这服务器端只是给回国用而已呢？兵不适合出口转内销用？
3：DDNS插件获得的IP和我实际链接的IP不一样，因为PCCW的入户是DHCP模式，所以IP貌似不对；

以上就是刚开始安装的问题了，感谢各位大神帮忙了！
感谢各位支持！

LeeMenHin, 2020-02-29:
我在香港可以打开这个网站

liruishengus, 2020-02-29:
> 我在香港可以打开这个网站

我也可以打开这个网站，但是只要经过openwrt之后就上不去，连ping都ping不通...
所以我怀疑是不是openwrt里面有限制了什么？

9l, 2020-03-01:
换香港运营商提供的dns

liruishengus, 2020-03-01:
> 换香港运营商提供的dns

我并没有设置手动DNS，都是使用内网DHCP，我都是选择了默认设置（openwrt即网关也是DNS）的...但就是这个1688网站用不到搜索功能...

closed 2020-03-06 by coolsnowwolf

junjiego, 2020-03-13:
> 2：看到很多插件，只看到SSR+里面有个服务器端设置，但是里面的加密/协议/混淆都只有一种两种，并没有那些机场常用的加密/协议/混淆的选项选择，是不是这服务器端只是给回国用而已呢？兵不适合出口转内销用？

如果你会编译固件的话，可以把v2ray-server编译进去，里面的加密/协议/混淆的选项比ssr里的服务器完整一些。

> 3：DDNS插件获得的IP和我实际链接的IP不一样，因为PCCW的入户是DHCP模式，所以IP貌似不对；

ddns的ip获取方式把从端口获取改为通过url获取。