coolsnowwolf/lede #3424: shadowsocks plus+是不是得考虑改名了

lloyd18, 2020-02-29:
都支持ss/ssr/v2ray/trrjan了，还用这个名字是不是多少有点不合适了，而且ss/ssr容易被封服务器IP，还是少碰为好吧

coolsnowwolf, 2020-02-29:
那叫啥？

JerryOrz, 2020-02-29:
proxy tools如何

loonyd, 2020-02-29:
叫kexueshangwang

Myoko, 2020-02-29:
那叫什么？

BlackTouma, 2020-02-29:
咋不叫红杏出墙呢？

lloyd18, 2020-02-29:
问叫啥真的是难到了，大家集思广益都来说说吧
我说几个 Tunnel=隧道，ladder=梯子，Connect world=连接世界 
或者直接P+，P可以代表pass通过 可以代表proxy代理 +嘛，可以理解为更好 就如英特尔14nm++++

coolsnowwolf, 2020-02-29:
其实是这样的，+ 号越多证明越厉害的感觉

不如改名叫做 SSR Plus++++ 如何

yinzhefuyun, 2020-02-29:
> 其实是这样的，+ 号越多证明越厉害的感觉
> 
> 不如改名叫做 SSR Plus++++ 如何

NB Plus +++++ (支持几种协议几个+)

frely, 2020-02-29:
Free Plus

hyird, 2020-02-29:
Turbo Over Wall 咋样

lqhorochi, 2020-02-29:
叫power proxy
或者叫AIO Proxy  (All In One proxy)

hyird, 2020-02-29:
“我要出去看奶子” 怎么样

coolsnowwolf, 2020-02-29:
叫 XX Proxy 的 ， 带 VPN 字样的，带 Wall 的，都死透了，不吉利

hyird, 2020-02-29:
Moving bricks  
要不搬砖？
把墙的砖搬了打洞

swxk521, 2020-02-29:
跟上时代步伐就叫COVID-19PLUS

diligentpig, 2020-02-29:
> There are only two hard things in Computer Science: cache invalidation and naming things.

感觉 @yinzhefuyun 的建议不错 

> > 其实是这样的，+ 号越多证明越厉害的感觉
> > 不如改名叫做 SSR Plus++++ 如何
> 
> NB Plus +++++ (支持几种协议几个+)

只是加号太多数不清楚，不如改成数字： NB Plus +3 (现在是ss, v2ray, trojan?)

lqhorochi, 2020-02-29:
干脆叫Leap plugin

Ryuchen, 2020-02-29:
我推荐男人的默契，韩国的那个PS4广告，直接叫Plash Speed

ch1993128, 2020-03-01:
ssr plus里面我之用trojan和v2ray。。。一用ss和ssr立马服务器ip被封

9l, 2020-03-01:
跟上开源的传统，librepass plus ++++++++++++++++++
还有 freepass plus+++++++++++++, vipass plus++++++++++++++++, let's pass plus++++++++++++++++++++

kl9800, 2020-03-01:
L'Arc〜en〜Ciel 哈哈哈哈哈哈哈

lonee6, 2020-03-01:
跟我出国看奶子++

jyy5206, 2020-03-01:
SSRVT

说明：
SS=SS
R=SSR
V=V2Ray
T=Trojan

好记

Mr. Lean 喜欢+的话：
SSRVT+

Plus 就不用了
因为 “Plus" = "+"

hcym, 2020-03-01:
飞雕吉利，屌炸天霸气

lloyd18, 2020-03-01:
> SSRVT
> 
> 说明：
> SS=SS
> R=SSR
> V=V2Ray
> T=Trojan
> 
> 好记
> 
> Mr. Lean 喜欢+的话：
> SSRVT+
> 
> Plus 就不用了
> 因为 “Plus" = "+"

下次添加个工具岂不是要再加个字母

greennyreborn, 2020-03-01:
不用改名，名字是个故事，可以流传

KFERMercer, 2020-03-01:
为这种 all in one 的工具命名并不应该倾向于其包含的任何功能, 也不应该过于直白的描述它的作用. 需要用一个新词来突出它的工具属性, 同时要兼具内涵.

在这里推荐一个名字: `SilverKey` 银匙

《银钥匙》是克苏鲁神话之父洛夫克拉夫特的一篇奇幻散文, 讲述了调查员伦道夫·卡特(Randolph Carter)借助一把祖传的银钥匙而习得了宇宙的秘密, 在窥见了宇宙的一斑之后都陷入了无尽的疯狂.

> 这篇小说可以说是最直观地描述了洛夫克拉夫特一直所说的那种“宏大的恐惧”的概念，但因为太过抽象在很多地方依然会引起读者的不解和误读——比如经常有人会产生疑问，那个疑似“犹格索托斯”的存在表现得太过于“友好”。但实际上在小说中洛夫克拉夫特明确地提出了那是一种超越了人类善恶观念的存在，一切凡人的世俗的存在——不论是表现出的憎恨也好、愤怒也好、博爱也好或者虚荣也好，又或者是那些赞美和献祭，对理性的或者非理性的，或者对于自然的崇拜和信仰与之相比都完全不值一提。

[《银钥匙》全文](https://coc-zh.jokester.io/HPL/The-Silver-Key.html)

jyy5206, 2020-03-01:
> 为这种 all in one 的工具命名并不应该倾向于其包含的任何功能, 也不应该过于直白的描述它的作用. 需要用一个新词来突出它的工具属性, 同时要兼具内涵.
> 
> 在这里推荐一个名字: `SilverKey` 银匙
> 
> 《银钥匙》是克苏鲁神话之父洛夫克拉夫特的一篇奇幻散文, 讲述了调查员伦道夫·卡特(Randolph Carter)借助一把祖传的银钥匙而习得了宇宙的秘密, 在窥见了宇宙的一斑之后陷入了无尽的疯狂.
> 
> > 这篇小说可以说是最直观地描述了洛夫克拉夫特一直所说的那种“宏大的恐惧”的概念，但因为太过抽象在很多地方依然会引起读者的不解和误读——比如经常有人会产生疑问，那个疑似“犹格索托斯”的存在表现得太过于“友好”。但实际上在小说中洛夫克拉夫特明确地提出了那是一种超越了人类善恶观念的存在，一切凡人的世俗的存在——不论是表现出的憎恨也好、愤怒也好、博爱也好或者虚荣也好，又或者是那些赞美和献祭，对理性的或者非理性的，或者对于自然的崇拜和信仰与之相比都完全不值一提。
> > https://www.gcores.com/articles/101414
> 
> [《银钥匙》全文](https://coc-zh.jokester.io/HPL/The-Silver-Key.html)

这里都是理科直男
需要你这样的文科生~

lifetyper, 2020-03-01:
SSR Plus Max ,妥妥的。

lin-xh, 2020-03-02:
SeeTheWorld

rampageX, 2020-03-02:
ASS HOLE XXXXXXXL, `X` 的数量随意

tuTuStream, 2020-03-03:
> 
> 
> 我推荐男人的默契，韩国的那个PS4广告，直接叫Plash Speed

Plash Speed 4 支持4种协议 完美

martin0607, 2020-03-03:
GFWgodie

Xr267, 2020-03-04:
Let's Plus

mjyhj, 2020-03-05:
叫mj即可

closed 2020-03-06 by coolsnowwolf