coolsnowwolf/lede #3426: 构建固件时出现问题。

butinhi, 2020-02-29:
Can't exec "curl": No such file or directory at /home/butinhi/lede/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --no-check-certificate --output-document=- https://github.com/upx/upx/releases/download/v3.95/upx-3.95-src.tar.xz
--2020-02-29 11:03:17--  https://github.com/upx/upx/releases/download/v3.95/upx-3.95-src.tar.xz
Resolving github.com (github.com)... 140.82.118.3
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/67031040/ead24180-a8f2-11e8-81dd-76617546824c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200229%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200229T110318Z&X-Amz-Expires=300&X-Amz-Signature=4b7180634ea26960935d70814d7bb5ecfc0f1746e6ce129ff19291b64d32297d&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dupx-3.95-src.tar.xz&response-content-type=application%2Foctet-stream [following]
--2020-02-29 11:03:19--  https://github-production-release-asset-2e65be.s3.amazonaws.com/67031040/ead24180-a8f2-11e8-81dd-76617546824c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200229%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200229T110318Z&X-Amz-Expires=300&X-Amz-Signature=4b7180634ea26960935d70814d7bb5ecfc0f1746e6ce129ff19291b64d32297d&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dupx-3.95-src.tar.xz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... failed: Name or service not known.
wget: unable to resolve host address 'github-production-release-asset-2e65be.s3.amazonaws.com'
Download failed.
Can't exec "curl": No such file or directory at /home/butinhi/lede/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --no-check-certificate --output-document=- https://sources.openwrt.org/upx-3.95-src.tar.xz
--2020-02-29 11:03:19--  https://sources.openwrt.org/upx-3.95-src.tar.xz
Resolving sources.openwrt.org (sources.openwrt.org)... 148.251.151.136
Connecting to sources.openwrt.org (sources.openwrt.org)|148.251.151.136|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-29 11:03:20 ERROR 404: Not Found.

Download failed.
Can't exec "curl": No such file or directory at /home/butinhi/lede/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --no-check-certificate --output-document=- https://mirror2.openwrt.org/sources/upx-3.95-src.tar.xz
--2020-02-29 11:03:20--  https://mirror2.openwrt.org/sources/upx-3.95-src.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-29 11:03:21 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Makefile:37: recipe for target '/home/butinhi/lede/dl/upx-3.95-src.tar.xz' failed
make[3]: *** [/home/butinhi/lede/dl/upx-3.95-src.tar.xz] Error 2
make[3]: Leaving directory '/home/butinhi/lede/tools/upx'
time: tools/upx/compile#0.06#0.05#4.09
tools/Makefile:161: recipe for target 'tools/upx/compile' failed
make[2]: *** [tools/upx/compile] Error 2
make[2]: Leaving directory '/home/butinhi/lede'
tools/Makefile:159: recipe for target '/home/butinhi/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnyyyyynyyyyyyyyyyyyyyynnynynnyyyynynyyy' failed
make[1]: *** [/home/butinhi/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnyyyyynyyyyyyyyyyyyyyynnynynnyyyynynyyy] Error 2
make[1]: Leaving directory '/home/butinhi/lede'
/home/butinhi/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

guochan2019, 2020-02-29:
拉代码错误吧，全程科学上网

butinhi, 2020-02-29:
> 拉代码错误吧，全程科学上网

我尝试了很多次，但仍然是相同的错误。

jzh33, 2020-02-29:
网络不行，下载错误

jzh33, 2020-02-29:
先完善编译环境，确保全程科学。。。你这是下载出错，有可能是下载工具也可能是网络

butinhi, 2020-02-29:
> 先完善编译环境，确保全程科学。。。你这是下载出错，有可能是下载工具也可能是网络
我仍然正常使用互联网，因此不清楚错误出在哪里。

liwenjie119, 2020-03-01:
https://mirror2.openwrt.org/sources/upx-3.95-src.tar.xz
这个网址404了，也就是在这没有这个文件了，需要其他的更新源。但是不清楚为什么github上的正常文件你没下下来，应该是科学有问题

closed 2020-03-06 by coolsnowwolf