coolsnowwolf/lede #3436: 三次编译都失败

TomNotchYmk, 2020-02-29:
[image]
编译机器配置：
谷歌云1核vps，香港C区（其实应该与这个没多大关系）
系统：Ubuntu 14.04 64位

此函数无缺省默认值，或者调用错误。

lamu0102, 2020-03-01:
阿里云c区不行啊。而且也不适合编译

AmadeusGhost, 2020-03-01:
不要选olsrd这个包，似乎上游源码有问题

TomNotchYmk, 2020-03-01:
> 阿里云c区不行啊。而且也不适合编译

我初次编译就是用的同一台机器，而且这明显不是阿里云的问题。

yinzhefuyun, 2020-03-02:
命令行输入 sudo apt-get update ，然后输入
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib linux-libc-dev:i386

closed 2020-03-06 by coolsnowwolf