coolsnowwolf/lede #3447: newifi3启用bbr导致usb口上传速度变慢

DongfeiSay, 2020-03-01:
网线直连，插的usb2.0u盘，关闭bbr20MB/s，开启bbr3.1MB/s

eekon, 2020-03-01:
drop it

467815891a, 2020-03-01:
bbr别开，我在7621上用Trojan，开BBR就8Mbps。BBR关掉直接30Mbps（到服务器上限了）

hcym, 2020-03-01:
还有这种事，不知道什么原理，

DongfeiSay, 2020-03-01:
> bbr别开，我在7621上用Trojan，开BBR就8Mbps。BBR关掉直接30Mbps（到服务器上限了）

问题是开了bbr 连openwrt的ss server能跑满百兆带宽，关了的话只能跑58左右

467815891a, 2020-03-01:
> > bbr别开，我在7621上用Trojan，开BBR就8Mbps。BBR关掉直接30Mbps（到服务器上限了）
> 
> 问题是开了bbr 连openwrt的ss server能跑满百兆带宽，关了的话只能跑58左右

没办法了，估计bbr只对服务端有奇效，你可以再买一台设备，比如nanopi之类的单独做服务器

DongfeiSay, 2020-03-01:
> > > bbr别开，我在7621上用Trojan，开BBR就8Mbps。BBR关掉直接30Mbps（到服务器上限了）
> > 
> > 
> > 问题是开了bbr 连openwrt的ss server能跑满百兆带宽，关了的话只能跑58左右
> 
> 没办法了，估计bbr只对服务端有奇效，你可以再买一台设备，比如nanopi之类的单独做服务器

主要是会连累到usb口就有点奇怪了

lunatickochiya, 2020-03-03:
已经有4.14内核的bbr解决方案了，但是还是等5.4内核的bbr v2比较实在

DongfeiSay, 2020-03-03:
> 已经有4.14内核的bbr解决方案了，但是还是等5.4内核的bbr v2比较实在

那是不知道有多久op内核才会更新到5.4了

lunatickochiya, 2020-03-03:
> > 已经有4.14内核的bbr解决方案了，但是还是等5.4内核的bbr v2比较实在
> 
> 那是不知道有多久op内核才会更新到5.4了

官方已经出了

467815891a, 2020-03-03:
x86也就有了，arm也在测试，但是要移植到MIPS还很早，目前hwnat都没有搞好

closed 2020-03-06 by coolsnowwolf