coolsnowwolf/lede #3451: 系统日志的SSR-Server 满屏输出No file descriptors available的错误

babylife, 2020-03-01:
貌似是最新版本的ssr才有问题，刷了2个不同编译作者的k3的最新版本openwrt的，都有这个问题，最新的版本更新了trojan的多线程支持，不知道是不是有关系。 L大看看？

Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available
Sun Mar  1 20:15:59 2020 kern.err /usr/bin/ssr-server[19846]: accept: No file descriptors available

babylife, 2020-03-01:
我是开启了ssr server给自己内网做链接的，报错虽然存在，Trojan正常，通过ssr server分享给其他客户端的功能也正常，就是系统日志一直输出这个日志

coolsnowwolf, 2020-03-01:
echo 65535 > /proc/sys/fs/file-max

babylife, 2020-03-01:
> echo 65535 > /proc/sys/fs/file-max

老大修改好了 还是老样子 我看默认值是51157，会不会是其他原因？
我的环境：第一个k3 刷openwrt最新 开启trojan节点和ssr servet。 另一个k3 刷梅林，软件中心的ssr链接第一个k3 做ssr代理。
目前功能全部正常就是系统日志疯狂满屏输出这个。 刷了两个版本都这样。

babylife, 2020-03-01:
[image]
[image]

babylife, 2020-03-02:
今天早上又是满屏报错。崩溃中
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: accept: No file descriptors available
[... cut ...]
Mon Mar  2 08:13:21 2020 kern.err /usr/bin/ssr-server[32630]: connect error
Mon Mar  2 08:13:23 2020 kern.err /usr/bin/ssr-server[32630]: socket: No file descriptors available
Mon Mar  2 08:13:23 2020 kern.err /usr/bin/ssr-server[32630]: socket: No file descriptors available
Mon Mar  2 08:13:23 2020 kern.err /usr/bin/ssr-server[32630]: connect error
Mon Mar  2 08:13:34 2020 daemon.err odhcp6c[1518]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Mon Mar  2 08:13:40 2020 daemon.err uhttpd[1763]: sed: write error

babylife, 2020-03-02:
我的ssr配置文件
root@Trojan:~# cat /var/etc/shadowsocksr_1.json
{
"server": "0.0.0.0",
"server_port": 8388,
"password": "phantasm",
"timeout": 60,
"method": "rc4-md5",
"protocol": "origin",
"protocol_param": "",
"obfs": "plain",
"obfs_param": "",
"fast_open": false
}

babylife, 2020-03-02:
L大，看看是不是你的代码，UI上保存的时候，漏掉了fast_open的值的引号引起的

KFERMercer, 2020-03-02:
布尔型不需要加引号

babylife, 2020-03-02:
> 布尔型不需要加引号

谢谢回复！

开启了ShadowsocksR Python 的ssr server之后，我另一个路由器连接这个就很正常，也很稳定。 不确定ShadowSocksR Plus+的服务端是不是有问题。 算是临时找到替代方法了。

coolsnowwolf, 2020-03-02:
建议用 完整版 的 ssr python server

babylife, 2020-03-03:
> 建议用 完整版 的 ssr python server

昨天又找了x86的32位固件，测试ssr server 貌似没有问题，也打算用那个软路由做trojan代理转发了，毕竟这样使用的人不多，先观望下吧，谢谢L大！

closed 2020-03-03 by babylife