coolsnowwolf/lede #3453: 编译失败，Deepin 没有linux-libc-dev:i386。

mcucore, 2020-03-01:
Deepin 没有linux-libc-dev:i386，编译失败，编译OPENWRT官方可以的。只好又换回了Ubuntu 14。

honjow, 2020-03-01:
我直接改成linux-libc-dev。反正我是编好的。。

smsource, 2020-03-01:
试一试 dpkg --add-architecture i386  这个命令

AmadeusGhost, 2020-03-02:
linux-libc-dev:i386是ubuntu14.04的包，18.04和deepin用下面的依赖：
```
sudo apt-get install -y build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libreadline-dev libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint ccache curl wget scons python python3  haveged lrzsz device-tree-compiler
```

mcucore, 2020-03-02:
> linux-libc-dev:i386是ubuntu14.04的包，18.04和deepin用下面的依赖：
>
> ```
> sudo apt-get install -y build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 [...]

感谢！又可以愉快的用DEEPIN了，虽然没弄懂那个32位依赖包。

closed 2020-03-02 by mcucore