coolsnowwolf/lede #3454: frpc穿透在部分设备上占用异常

DongfeiSay, 2020-03-01:
使用设备：newifi3 mt7621a 版本都是0.31.2.，不知其他mt7621a机型是否会有这个问题
[image]
[image]
如图，newifi3的frpc进程VSZ高达656m，%VSZ高达131%？  86的可以看到是正常的占用情况

另外协议类型选择kcp甚至会将Load average负载均衡值将长久增加0.4~1(1-1.6徘徊)，x86的也会增加了0.2~0.3，是否正常？
[image]

DongfeiSay, 2020-03-01:
> 使用设备：newifi3 mt7621a版本都是0.31.2.，不知其他mt7621a机型是否会有这个问题 如图，newifi3的frpc进程VSZ高达656m，%VSZ高达131%？86的可以看到是正常的占用情况
> [image]
> [image]
> 
> 另外协议类型选择kcp甚至会将Load average负载均衡值将长久增加0.4-1(1-1.6徘徊)，x86的也会增加了0.2-0.3，是否正常？
> [image]

coolsnowwolf, 2020-03-01:
这些没有 fpu 的 处理器靠模拟运行的，占用都很高

uoow, 2020-03-04:
> 这些没有 fpu 的 处理器靠模拟运行的，占用都很高

[image]
L大，请问FRP这个错误提示是不是bug？1月的版本使用是正常的，回来编译新版本后，FRP就连不了；确认过服务端与路由的多路复用设置是一样的。

closed 2020-03-06 by coolsnowwolf