coolsnowwolf/lede #3457: 绿联2.5Gbps的USB网卡无法被识别为eth设备

crosys, 2020-03-01:
大大，反复试过，就是没法驱动绿联的U口2.5G网卡（硬件id是0bda:8156），手边正好还有一个可以正常驱动的8152的网卡，对比了一下插上去之后系统的提示，似乎是无法把8156识别为eth设备所致。

看起来似乎集成的还是1.09.9的驱动，不能有没有办法替换为官网上2.12.0的版本(https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software)，那个应该可以支持8156了。

root@OpenWrt:~# lsusb
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 0bda:8156 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

coolsnowwolf, 2020-03-02:
加了官方驱动，但不知道是否能用，没设备

crosys, 2020-03-02:
> 加了官方驱动，但不知道是否能用，没设备

太给力了，晚上就试一下，顺便请教一下，是kmod-usb-net-rtl8152更新了，还是需要重新编译？

hcym, 2020-03-02:
软路由用usb真的强强联合没意思，我准备等内核上5.4以后搞搞ax200，电脑linux用的真好，还便宜啊，都是穷逼害的观点😂

crosys, 2020-03-03:
> 加了官方驱动，但不知道是否能用，没设备

昨天到家太晚，尝试编译时没看到8152/8156的选项就没折腾。上午看到恩山有基于昨天源码新编译的固件，试了一下，已经能驱动了。不过公司没有2.5G交换机，晚上回家才能再试是否支持2.5G，会再来反馈。估计问题不大，万分感谢大大！

crosys, 2020-03-04:
已确认，可以正常跑在2500M全双工模式下，完美解决问题，万分感谢！ @coolsnowwolf

crosys, 2020-03-31:
@L大，R20.3.19之后源码编译的固件又都无法驱动RTL8152网卡了，不知道有没有办法解决？或是是否需要手工安装某个软件包？

coolsnowwolf, 2020-04-08:
因为那个和 USB 的 815X 千兆系列冲突，正在想办法

你有这个网卡的购买链接吗

crosys, 2020-04-09:
> 因为那个和 USB 的 815X 千兆系列冲突，正在想办法
> 
> 你有这个网卡的购买链接吗

京东自营还没有，只有第三方店铺的：
https://item.jd.com/62531315014.html

天猫官方旗舰店的：
https://detail.tmall.com/item.htm?id=607985548355&skuId=4272714473099

另外，因为是type-c接口的，如果电脑没C口，就还需要一个A转C的转接头。

cookier0218, 2020-04-11:
目前编译了8150及8152
测试辨识不到8156网卡,
请问有其他要设定？
或是取消8150, 可以避免冲突的问题?

RayleiLi, 2020-04-12:
我也有同样的问题，2.5Gbps设备插上不能识别为网口了

RayleiLi, 2020-04-12:
> 目前编译了8150及8152
> 测试辨识不到8156网卡,
> 请问有其他要设定？
> 或是取消8150, 可以避免冲突的问题?

请问编译的时候如何选择8156的驱动呀？我找了好久都没找到，只找到了8152的。

RayleiLi, 2020-04-12:
> > 因为那个和 USB 的 815X 千兆系列冲突，正在想办法
> > 你有这个网卡的购买链接吗
> 
> 京东自营还没有，只有第三方店铺的：
> https://item.jd.com/62531315014.html
> 
> 天猫官方旗舰店的：
> https://detail.tmall.com/item.htm?id=607985548355&skuId=4272714473099
> 
> 另外，因为是type-c接口的，如果电脑没C口，就还需要一个A转C的转接头。

请问编译8156的驱动需要打开哪些开关呀？谢谢你。

crosys, 2020-04-12:
> > > 因为那个和 USB 的 815X 千兆系列冲突，正在想办法
> > > 你有这个网卡的购买链接吗
> > 
> > 
> > 京东自营还没有，只有第三方店铺的：
> > https://item.jd.com/62531315014.html
> > 天猫官方旗舰店的：
> > https://detail.tmall.com/item.htm?id=607985548355&skuId=4272714473099
> > 另外，因为是type-c接口的，如果电脑没C口，就还需要一个A转C的转接头。
> 
> 请问编译8156的驱动需要打开哪些开关呀？谢谢你。

C大还没解决冲突问题，要么用比较老的内核（确认4.19.108正常支持8156），要么等后续更新。

RayleiLi, 2020-04-12:
> > > > 因为那个和 USB 的 815X 千兆系列冲突，正在想办法
> > > > 你有这个网卡的购买链接吗
> > > 
> > > 
> > > 京东自营还没有，只有第三方店铺的：
> > > https://item.jd.com/62531315014.html
> > > 天猫官方旗舰店的：
> > > https://detail.tmall.com/item.htm?id=607985548355&skuId=4272714473099
> > > 另外，因为是type-c接口的，如果电脑没C口，就还需要一个A转C的转接头。
> > 
> > 
> > 请问编译8156的驱动需要打开哪些开关呀？谢谢你。
> 
> C大还没解决冲突问题，要么用比较老的内核（确认4.19.108正常支持8156），要么等后续更新。

嗯嗯，我现在试试老的版本，但是不知道编译需要开什么开关。所以想请教你一下。

crosys, 2020-04-12:
> 嗯嗯，我现在试试老的版本，但是不知道编译需要开什么开关。所以想请教你一下。

我没有实际编译过，都是直接用的别人编译好的版本。因为当时C大把驱动集成后，紧跟着恩山的好几个固件就都能支持了，应该是那段时间的内核默认编译设置就能支持。

RayleiLi, 2020-04-12:
> > 嗯嗯，我现在试试老的版本，但是不知道编译需要开什么开关。所以想请教你一下。
> 
> 我没有实际编译过，都是直接用的别人编译好的版本。因为当时C大把驱动集成后，紧跟着恩山的好几个固件就都能支持了，应该是那段时间的内核默认编译设置就能支持。

谢谢你的回复。

crosys, 2020-04-12:
> > > 嗯嗯，我现在试试老的版本，但是不知道编译需要开什么开关。所以想请教你一下。
> > 
> > 
> > 我没有实际编译过，都是直接用的别人编译好的版本。因为当时C大把驱动集成后，紧跟着恩山的好几个固件就都能支持了，应该是那段时间的内核默认编译设置就能支持。
> 
> 谢谢你的回复。

C大之前3月2日加入的新驱动，什么时候移除的不确定。应该用3月3日到大约3月15日编译的固件都可以支持8156，不过对8150的支持可能就有问题了。

azazjjyjjy, 2020-04-20:
最近正在考虑买这个USB网卡，楼主你后面打算怎么用？4个这个组成两对？

RayleiLi, 2020-04-20:
对的，这个必须得一对一对得才行。

crosys, 2020-04-22:
> 最近正在考虑买这个USB网卡，楼主你后面打算怎么用？4个这个组成两对？

有支持不算太贵的交换机可以买到了，亚马逊海外购买MS510TX或是阿里巴巴买TL-SH1206。

RayleiLi, 2020-05-24:
这个2.5G的USB网卡太好用了

RayleiLi, 2020-05-24:
> > 最近正在考虑买这个USB网卡，楼主你后面打算怎么用？4个这个组成两对？
> 
> 有支持不算太贵的交换机可以买到了，亚马逊海外购买MS510TX或是阿里巴巴买TL-SH1206。

请问你知道是否有2.5Gbps网口的光猫用于替换宽带家用的电信光猫吗？我把宽带升级到了1000M感觉赠送的光猫无法跑满。

crosys, 2020-06-23:
> > > 最近正在考虑买这个USB网卡，楼主你后面打算怎么用？4个这个组成两对？
> > 
> > 
> > 有支持不算太贵的交换机可以买到了，亚马逊海外购买MS510TX或是阿里巴巴买TL-SH1206。
> 
> 请问你知道是否有2.5Gbps网口的光猫用于替换宽带家用的电信光猫吗？我把宽带升级到了1000M感觉赠送的光猫无法跑满。

万兆wan，2.5G lan的光猫华为出了(搜optixstar 光猫），之前说517电信日会跟着电信2000M宽带一起发布，但是目前还没有看到。。。
除此之外就只有中兴F1400（4Base-T口的也不行，有lan-xg的SFP+口的才可以），或炒上天的华为8055Q了。
我自己用的是F1400，配合刷@coolsnowwolf大固件的软路由，电信千兆宽带测速能跑到1200的样子，比电信送的sdn网关能提速差不多1/4。
你要换的话，需要特别注意自己所在地区是否光猫配了正确的loid就能用，那就能自己搞定；否则就要找运营商的人帮忙了。

RayleiLi, 2020-06-25:
> 万兆wan，2.5G lan的光猫华为出了(搜optixstar 光猫），之前说517电信日会跟着电信2000M宽带一起发布，但是目前还没有看到。。。
>
> 除此之外就只有中兴F1400（4Base-T口的也不行，有lan-xg的SFP+口的才可以），或炒上天的华为8055Q了。 [...]

好的，谢谢你。上海电信似乎不可以自己换光猫，我这周末就去咨询一下。现在家里最高也只能跑到1000多一点点。

RayleiLi, 2020-06-25:
> > > > > > 最近正在考虑买这个USB网卡，楼主你后面打算怎么用？4个这个组成两对？
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > 有支持不算太贵的交换机可以买到了，亚马逊海外购买MS510TX或是阿里巴巴买TL-SH1206。
> 
> > > 
> 
> > > 
> 
> > > > 
> 
> > > 
> 
> > > 
> 
> > > > 请问你知道是否有2.5Gbps网口的光猫用于替换宽带家用的电信光猫吗？我把宽带升级到了1000M感觉赠送的光猫无法跑满。
> 
> > > 
> 
> > > 
> 
> > > 万兆wan，2.5G lan的光猫华为出了(搜optixstar 光猫），之前说517电信日会跟着电信2000M宽带一起发布，但是目前还没有看到。。。
> 
> > > 除此之外就只有中兴F1400（4Base-T口的也不行，有lan-xg的SFP+口的才可以），或炒上天的华为8055Q了。
> 
> > > 我自己用的是F1400，配合刷@coolsnowwolf大固件的软路由，电信千兆宽带测速能跑到1200的样子，比电信送的sdn网关能提速差不多1/4。
> 
> > > 你要换的话，需要特别注意自己所在地区是否光猫配了正确的loid就能用，那就能自己搞定；否则就要找运营商的人帮忙了。
> 
> > 
> 
> > 好的，谢谢你。上海电信似乎不可以自己换光猫，我这周末就去咨询一下。现在家里最高也只能跑到1000多一点点。
> 
> 
> 
> 魔都绝大部分区域都可以，可以找个光猫试试，连上当前的sdn网关记下来loid，配置到要换的光猫里就行了，密码随意填

谢谢你，你很专业哦。我们这个片区的电信师傅很配合，每次问他问题都帮忙解决。我想等一下你说的那个新的华为光猫再换，那个光猫有个2.5G的lan口。似乎不错。

crosys, 2020-06-26:
> > 魔都绝大部分区域都可以，可以找个光猫试试，连上当前的sdn网关记下来loid，配置到要换的光猫里就行了，密码随意填
> 
> 谢谢你，你很专业哦。我们这个片区的电信师傅很配合，每次问他问题都帮忙解决。我想等一下你说的那个新的华为光猫再换，那个光猫有个2.5G的lan口。似乎不错。

不急的话确实可以等魔都2000M宽带套餐，估计会免费换华为这款光猫（不然网速肯定不达标），即使升级2000M宽带有差价，估计也比自己买万兆光猫要划算。

RayleiLi, 2020-06-27:
> > > 魔都绝大部分区域都可以，可以找个光猫试试，连上当前的sdn网关记下来loid，配置到要换的光猫里就行了，密码随意填
> 
> > 
> 
> > 谢谢你，你很专业哦。我们这个片区的电信师傅很配合，每次问他问题都帮忙解决。我想等一下你说的那个新的华为光猫再换，那个光猫有个2.5G的lan口。似乎不错。
> 
> 
> 
> 不急的话确实可以等魔都2000M宽带套餐，估计会免费换华为这款光猫（不然网速肯定不达标），即使升级2000M宽带有差价，估计也比自己买万兆光猫要划算。

好的👌，你说的有道理。

closed 2020-07-04 by coolsnowwolf

liuquan321, 2020-12-31:
@coolsnowwolf lean大佬目前21年1月的源码还是无法驱动8156和8156b，如果之前的做法会和千兆网卡冲突，能不能考虑在其他编译条目里留一条可选的，让有需求的人二选一？