coolsnowwolf/lede #3459: 今天2.4g信号不正常，经常丢失或突然拒绝连接(固件是2.29晚编译的)

abclevel, 2020-03-01:
今天2.4g信号不正常，经常丢失或突然拒绝连接(固件是2.29晚编译的)

Aes64X, 2020-03-01:
型号？日志？过程？开局一句话，剩下你让谁编？

abclevel, 2020-03-02:
主机名 | R8000
-- | --
主机型号 | Netgear R8000 (BCM4709) (CpuMark : 7301.768397 Scores)
架构 | ARMv7 Processor rev 0 (v7l) x 2
固件版本 | OpenWrt R20.2.15 / LuCI Master (git-20.058.13156-2faf063)
内核版本 | 4.14.171

Mon Mar  2 00:02:01 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: associated
Mon Mar  2 00:02:02 2020 daemon.notice hostapd: wlan1: AP-STA-POSSIBLE-PSK-MISMATCH 90:94:97:7a:09:e4
Mon Mar  2 00:02:05 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:05 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:05 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: associated
Mon Mar  2 00:02:09 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:09 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:09 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: associated
Mon Mar  2 00:02:13 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:13 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:20 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:02:22 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: associated
Mon Mar  2 00:02:22 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 90:94:97:7a:09:e4
Mon Mar  2 00:02:22 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 WPA: pairwise key handshake completed (WPA)
Mon Mar  2 00:02:23 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 WPA: group key handshake completed (WPA)
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:04:57 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 90:94:97:7a:09:e4
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan1: STA 90:94:97:7a:09:e4 IEEE 802.11: disassociated
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan0: STA 90:94:97:7a:09:e4 IEEE 802.11: associated
Mon Mar  2 00:04:57 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 90:94:97:7a:09:e4
Mon Mar  2 00:04:57 2020 daemon.info hostapd: wlan0: STA 90:94:97:7a:09:e4 WPA: pairwise key handshake completed (WPA)
Mon Mar  2 00:04:58 2020 daemon.info hostapd: wlan0: STA 90:94:97:7a:09:e4 WPA: group key handshake completed (WPA)
Mon Mar  2 00:05:01 2020 kern.err /usr/bin/ssr-redir[6484]: server recv: Connection reset by peer
Mon Mar  2 00:07:54 2020 kern.info kernel: [ 2467.850972] device wlan1 left promiscuous mode
Mon Mar  2 00:07:54 2020 kern.info kernel: [ 2467.855631] br-lan: port 2(wlan1) entered disabled state
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 48:a1:95:77:4f:3b
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 98:cf:53:53:68:a5
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED b2:59:47:07:01:37
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED b2:59:47:09:b6:d2
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 50:ec:50:8c:63:b1
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 78:11:dc:b4:1d:b9
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 78:11:dc:fc:44:44
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Mon Mar  2 00:07:54 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Mon Mar  2 00:07:55 2020 kern.err kernel: [ 2468.409782] ieee80211 phy1: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
Mon Mar  2 00:07:55 2020 daemon.notice netifd: radio1 (11771): command failed: No error information (-524)
Mon Mar  2 00:07:55 2020 user.notice mac80211: Failed command: iw dev wlan1 del
Mon Mar  2 00:07:55 2020 daemon.notice netifd: radio1 (11892): command failed: No error information (-524)
Mon Mar  2 00:07:55 2020 user.notice mac80211: Failed command: iw dev wlan1 del
Mon Mar  2 00:07:55 2020 user.notice mac80211: Failed command: iw phy phy1 set antenna 0xffffffff 0xffffffff
Mon Mar  2 00:07:55 2020 daemon.notice netifd: radio1 (11892): command failed: I/O error (-5)
Mon Mar  2 00:07:55 2020 user.notice mac80211: Failed command: iw phy phy1 set distance 0
Mon Mar  2 00:07:56 2020 daemon.notice netifd: radio1 (11892): command failed: Too many open files in system (-23)
Mon Mar  2 00:07:56 2020 user.notice mac80211: Failed command: iw phy phy1 interface add wlan1 type __ap
Mon Mar  2 00:07:56 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Mon Mar  2 00:07:56 2020 kern.info kernel: [ 2469.381423] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Mon Mar  2 00:07:56 2020 kern.info kernel: [ 2469.394257] br-lan: port 2(wlan1) entered blocking state
[... cut ...]
Mon Mar  2 01:44:26 2020 kern.info /usr/bin/ssr-redir[30436]: TCP relay disabled
Mon Mar  2 01:44:26 2020 kern.info /usr/bin/ssr-redir[30436]: listening at 0.0.0.0:1234
Mon Mar  2 01:44:26 2020 kern.info /usr/bin/ssr-redir[30436]: running from root user
Mon Mar  2 01:44:26 2020 daemon.info pdnsd[30452]: pdnsd-1.2.9b-par starting.
Mon Mar  2 01:44:27 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Mar  2 01:44:27 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

Aes64X, 2020-03-02:
大概率无线驱动的问题，暂时无解，我没找到其他的BCM4365驱动，貌似只能用op给的，不想4366有一大堆提取驱动。

abclevel, 2020-03-02:
> 大概率无线驱动的问题，暂时无解，我没找到其他的BCM4365驱动，貌似只能用op给的，不想4366有一大堆提取驱动。

好的，谢谢回复!

closed 2020-03-02 by coolsnowwolf