coolsnowwolf/lede #3462: 各位大佬求救3天了 无数次编译

TendernessRadif, 2020-03-02:
mv -f /home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_arm64/main /home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_arm64/v2ray
/home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/host/bin/upx --lzma --best /home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_arm64/v2ray
bash: /home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/host/bin/upx: No such file or directory
Makefile:307: recipe for target '/home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.22.1/.built' failed
make[3]: *** [/home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.22.1/.built] Error 127
make[3]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt/package/lean/v2ray'
time: package/lean/v2ray/compile#102.74#24.94#39.84
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt'
package/Makefile:107: recipe for target '/home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt'
/home/lbx/friendlywrt-rk3328/friendlywrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
Command exited with non-zero status 2
you take 1:10:39 to build friendlywrt
求大佬们支招

KFERMercer, 2020-03-02:
git clean -xdf && git pull

TendernessRadif, 2020-03-02:
老哥 完事之后是正常重来操作吗？

shu0802, 2020-03-02:
还是检查你的网络吧，应该是出在网络问题上

TendernessRadif, 2020-03-02:
> 还是检查你的网络吧，应该是出在网络问题上

老哥 应该不至于吧 全局状态 GCP 能37万速度，不是网络问题吧。。。。。

KFERMercer, 2020-03-02:
工具链没更新. 删掉重来就好了

TendernessRadif, 2020-03-02:
谢谢老哥 我再尝试下

shu0802, 2020-03-02:
关键要稳定，不能在某个时间出问题。这个与看视频还是有区别的，看视频可以缓冲，但编译是不能缓冲的哟。只要某个时间段网络出问题，都会出各种错误的。若网络不稳定，编译时得一定要关注到这个网络。

TendernessRadif, 2020-03-02:
> 关键要稳定，不能在某个时间出问题。这个与看视频还是有区别的，看视频可以缓冲，但编译是不能缓冲的哟。只要某个时间段网络出问题，都会出各种错误的。若网络不稳定，编译时得一定要关注到这个网络。

我明白您的意思了 谢谢哈 我感觉我要废了

you-mu, 2020-03-02:
编译的环境关掉ipv6,避免影响.
错误解决不了,就试试make clean.

TendernessRadif, 2020-03-02:
> 编译的环境关掉ipv6,避免影响.
> 错误解决不了,就试试make clean.

肯定关闭了ipv6 3天花了GCP 50刀 就是不成功 心累

jzh33, 2020-03-02:
你这是给R2S编译固件吧。。。我之前也是卡在v2ray了，取消掉upx压缩可以编译过去，你可以试试

TendernessRadif, 2020-03-02:
> 你这是给R2S编译固件吧。。。我之前也是卡在v2ray了，取消掉upx压缩可以编译过去，你可以试试

感谢老哥 ！！！但是upx压缩咋取消啊~~~~

YYiiEt, 2020-03-02:
我记得上次编译官方openwrt也没有upx，好像是从大雕源码里复制了一份工具过去解决了

iyuangang, 2020-03-03:
1. 将 L大 tools 下的 ucl upx 两个目录拷贝到 openwrt/tools
2. 编辑 openwrt/tools 下到Makefile
增加两行
```
tools-y += ucl upx
$(curdir)/upx/compile := $(curdir)/ucl/compile
```
3. make download -j8 V=s
4. make -j4 V=s || make -j1 V=s
应该解决了。
如果解决不了，可以参考我的openwrt 19.07.1 项目 [openwrt stable with lean's package](https://github.com/iyuangang/openwrt)

TendernessRadif, 2020-03-03:
> 1. 将 L大 tools 下的 ucl upx 两个目录拷贝到 openwrt/tools
> 2. 编辑 openwrt/tools 下到Makefile
>    增加两行
> 
> ```
> tools-y += ucl upx
> $(curdir)/upx/compile := $(curdir)/ucl/compile
> ```
> 
> 1. make download -j8 V=s
> 2. make -j4 V=s || make -j1 V=s
>    应该解决了。
>    如果解决不了，可以参考我的openwrt 19.07.1 项目 [openwrt stable with lean's package](https://github.com/iyuangang/openwrt)

老哥 我尝试了 问题依旧哈哈 谢谢您了

sofair, 2020-03-03:
` /home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/host/bin/upx: No such file or directory`
这句写的很清楚，缺少upx，可以从官网下个对应的 upx 放到上面的位置。可以用这条命令：`
 wget -q -O - https://github.com/upx/upx/releases/download/v3.96/upx-3.96-amd64_linux.tar.xz | tar -Jx --strip 1 -f - -C  /home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/host/bin upx-3.96-amd64_linux/upx`

closed 2020-03-06 by coolsnowwolf

fanck0605, 2020-04-13:
@TendernessRadif 参考这个 commit [af6e16](https://github.com/Lienol/openwrt/commit/af6e1632cd9250c0b4dc47606c1eb92ba4b3ae81)

snowrice, 2020-09-21:
> 1. 将 L大 tools 下的 ucl upx 两个目录拷贝到 openwrt/tools
> 2. 编辑 openwrt/tools 下到Makefile
>    增加两行
>
> ```
> tools-y += ucl upx
> $(curdir)/upx/compile := $(curdir)/ucl/compile
> ```
>
> 1. make download -j8 V=s
> 2. make -j4 [...]

大神牛B,真的可以了