coolsnowwolf/lede #3464: 新三多次尝试编译出错

lgfyxx, 2020-03-02:
代码：make[5]: *** [/home/runner/work/lede/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
make[4]: *** [node] Error 2
make[3]: *** [/home/runner/work/lede/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built] Error 2
make[2]: *** [package/feeds/packages/node/compile] Error 2
Makefile:101: recipe for target 'node' failed
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2
make[4]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/runner/work/lede/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built' failed
make[3]: Leaving directory '/home/runner/work/lede/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.26#0.10#0.36
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: Leaving directory '/home/runner/work/lede/lede'
package/Makefile:107: recipe for target '/home/runner/work/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.
不知道啥原因

closed 2020-03-02 by coolsnowwolf

lgfyxx, 2020-03-02:
有大佬帮忙解答一下吗

sundog315, 2020-03-14:
同求

rookiesky, 2020-05-08:
同求