coolsnowwolf/lede #3465: R3P 2.4g信号不正常。wifi 有时候一小时断2次

mingxiaoyu, 2020-03-02:
主机型号 | Xiaomi Mi Router 3 Pro
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt R20.2.15 / LuCI Master (git-20.058.13156-2faf063)
内核版本 | 4.14.171

Mon Mar  2 11:09:06 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 11:09:06 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 11:09:06 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:06 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 11:09:06 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:06 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 11:09:07 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 11:09:07 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:10 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 11:09:10 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 11:09:10 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:10 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 11:09:10 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 11:09:10 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:10 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 11:09:10 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:10 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 11:09:11 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 11:09:11 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:13 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 11:09:13 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 11:09:13 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:13 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 11:09:13 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 11:09:18 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:18 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 11:09:18 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 11:09:18 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:18 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 11:09:18 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 11:09:19 2020 daemon.info dnsmasq-dhcp[7373]: DHCPDISCOVER(br-lan) 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:19 2020 daemon.info dnsmasq-dhcp[7373]: DHCPOFFER(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:19 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 11:09:19 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 11:09:38 2020 daemon.info hostapd: wlan0: STA 1c:78:39:02:2f:20 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 11:09:38 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 1c:78:39:02:2f:20
Mon Mar  2 11:09:40 2020 daemon.info hostapd: wlan0: STA 1c:78:39:02:2f:20 IEEE 802.11: authenticated
Mon Mar  2 11:09:40 2020 daemon.info hostapd: wlan0: STA 1c:78:39:02:2f:20 IEEE 802.11: associated (aid 6)
Mon Mar  2 11:09:40 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 1c:78:39:02:2f:20
Mon Mar  2 11:09:40 2020 daemon.info hostapd: wlan0: STA 1c:78:39:02:2f:20 RADIUS: starting accounting session 2D7F13D38FA33EA4

Aes64X, 2020-03-02:
D1、R3G未见此问题，检查你的固件或设备

liuran001, 2020-03-02:
给你参考下
我r3g之前2.4g断流，5g打不开  
后来在无意间看到这个方法：
把2.4g改11信道
5g改149
地区代码全改成U.S.
然后重启就正常了

mingxiaoyu, 2020-03-02:
我2.4G  3 信道
5g  48 xindao
地区 澳大利亚

就是有时候 2.4g 就突然断了。。。。

mingxiaoyu, 2020-03-02:
Mon Mar  2 13:01:58 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 13:01:58 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:00 2020 cron.info crond[7515]: USER root pid 12221 cmd nohup /usr/share/passwall/monitor.sh > /dev/null 2>&1
Mon Mar  2 13:02:01 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 13:02:01 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 13:02:01 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:01 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 13:02:01 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 13:02:01 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:01 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:01 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:01 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:02 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 13:02:02 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:04 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 13:02:04 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 13:02:04 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:04 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 13:02:04 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 13:02:04 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:04 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:07 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 13:02:07 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:08 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 13:02:08 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 13:02:08 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:08 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 13:02:08 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 13:02:08 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:08 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:11 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:11 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:13 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:13 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:14 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Mar  2 13:02:14 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:15 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Mon Mar  2 13:02:15 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 5)
Mon Mar  2 13:02:16 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:16 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session 61B7E64B6D2E3159
Mon Mar  2 13:02:16 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Mon Mar  2 13:02:16 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:16 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:19 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:19 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:21 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:21 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:24 2020 daemon.info dnsmasq-dhcp[7373]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Mon Mar  2 13:02:24 2020 daemon.info dnsmasq-dhcp[7373]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Mon Mar  2 13:02:26 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs

又掉 一次

closed 2020-03-06 by coolsnowwolf