coolsnowwolf/lede #3468: 可否增加QNA-UCG51T的usb外置网卡的ACQ111U驱动，内含链接。

int2e, 2020-03-02:
usb3.0的5G/2.5G/1G的自适应网卡，我现在接到软路由上无法识别。

驱动下载地址：https://driverdownloads.aquantia.com/

[image]

hyird, 2020-03-02:
4.4 那没得办法了吧   还不如自己换个网卡

int2e, 2020-03-02:
我在内核模块里的usb驱动里把 usb-net下全部打钩。
然后usb3打钩，现在n能识别成下面这样，但是没法用，线插着的
[image]

`
[    0.000000] Linux version 4.19.106 (kunge@ubuntu1404) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2091-691def3)) #0 SMP Sat Feb 29 08:28:07 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=831342b2-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000008b1f9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000008b1fa000-0x000000008c06afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008c06b000-0x000000008c1d7fff] usable
[    0.000000] BIOS-e820: [mem 0x000000008c1d8000-0x000000008c599fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008c59a000-0x000000008cc0efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008cc0f000-0x000000008cc0ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008cc10000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Default string Default string/Default string, BIOS 5.13 08/21/2019
[    0.000000] tsc: Detected 2000.000 MHz processor
[    0.002818] tsc: Detected 1992.000 MHz TSC
[    0.002818] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.002821] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.002829] last_pfn = 0x46e000 max_arch_pfn = 0x400000000
[    0.002833] MTRR default type: write-back
[    0.002834] MTRR fixed ranges enabled:
[    0.002835]   00000-9FFFF write-back
[    0.002836]   A0000-BFFFF uncachable
[    0.002837]   C0000-FFFFF write-protect
[    0.002838] MTRR variable ranges enabled:
[    0.002839]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.002841]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.002841]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.002842]   3 base 008E000000 mask 7FFE000000 uncachable
[    0.002843]   4 base 008D800000 mask 7FFF800000 uncachable
[    0.002844]   5 base 2000000000 mask 6000000000 uncachable
[    0.002845]   6 base 1000000000 mask 7000000000 uncachable
[    0.002846]   7 base 0800000000 mask 7800000000 uncachable
[    0.002847]   8 base 4000000000 mask 4000000000 uncachable
[    0.002848]   9 disabled
[    0.004417] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.005385] last_pfn = 0x8cc10 max_arch_pfn = 0x400000000
[    0.015669] found SMP MP-table at [mem 0x000fcdc0-0x000fcdcf]
[    0.015685] Using GB pages for direct mapping
[    0.015688] BRK [0x02401000, 0x02401fff] PGTABLE
[    0.015690] BRK [0x02402000, 0x02402fff] PGTABLE
[    0.015691] BRK [0x02403000, 0x02403fff] PGTABLE
[    0.015759] BRK [0x02404000, 0x02404fff] PGTABLE
[    0.016323] BRK [0x02405000, 0x02405fff] PGTABLE
[    0.016420] BRK [0x02406000, 0x02406fff] PGTABLE
[    0.016615] ACPI: Early table checksum verification disabled
[    0.016668] ACPI: RSDP 0x00000000000F05B0 000024 (v02 ALASKA)
[... 636 lines cut ...]
[    6.563300] kmodloader: done loading kernel modules from /etc/modules.d/*
[    7.764620] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[    7.764620] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[    7.924154] Generic PHY r8169-100:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=r8169-100:00, irq=IGNORE)
[    7.937856] sdhci-pci 0000:00:14.5: SDHCI controller found [8086:9df5] (rev 30)
[    8.075276] r8169 0000:01:00.0 eth0: No native access to PCI extended config space, falling back to CSI
[    8.078560] br-lan: port 1(eth0) entered blocking state
[    8.079066] br-lan: port 1(eth0) entered disabled state
[    8.079706] device eth0 entered promiscuous mode
[    8.080669] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   10.719957] ksmbd: kill_server_store:480: kill command received
[   10.740345] ksmbd: create_socket:462: Failed to bind socket: -98
[   10.740917] ksmbd: tcp_destroy_socket:404: Failed to shutdown socket: -107
[... cut ...]
[   11.194041] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   11.559439] ax88179_178a 2-4:1.0 eth1: ax88179 - Link status is: 1
[   12.239041] ksmbd: kill_server_store:480: kill command received
[   16.911735] pppoe-wan: renamed from ppp0
[   64.757508] i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.
[   64.758554] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915

`

int2e, 2020-03-02:
> 4.4 那没得办法了吧 还不如自己换个网卡

4.2也支持，但是我们这个内核我看是4.19？

int2e, 2020-03-02:
@coolsnowwolf 驱动支持4.14内核，可以找个时间加下吗？谢谢！

coolsnowwolf, 2020-03-02:
没卡测试没法弄。。。

int2e, 2020-03-02:
> 没卡测试没法弄。。。

要不我邮寄给您，您测试好在发给我？运费我出，您给我留个地址就行。

int2e, 2020-03-02:
> 没卡测试没法弄。。。

int2e, 2020-03-02:
点错了，点成关闭问题了。
或者您给我留个地址，我淘宝买一个发给您，您留着用也行，没事的。都行。

closed 2020-03-02 by int2e

gzelvis, 2020-03-19:
想知道现在有openwrt的AQC111U的驱动了吗？我刚好也买了这个USB网卡，谢谢

gzelvis, 2020-03-25:
> usb3.0的5G/2.5G/1G的自适应网卡，我现在接到软路由上无法识别。
> 
> 
> 
> [补充：我在官网[产品规格](https://www.qnap.com/zh-cn/product/qna-uc5g1t/specs/hardware)看到这个，好像和当前内核版本不一致？是否可以尝试下能不能用？
> 
> [image]
> 
> 
> 
> 
> 
> 驱动下载地址：https://driverdownloads.aquantia.com/
> 
> 
> 
> [image]
> 
> 
> 
> 

请问你现在找到办法在openwrt下驱动aqc111u了吗

int2e, 2020-03-29:
> > usb3.0的5G/2.5G/1G的自适应网卡，我现在接到软路由上无法识别。
> > [补充：我在官网[产品规格](https://www.qnap.com/zh-cn/product/qna-uc5g1t/specs/hardware)看到这个，好像和当前内核版本不一致？是否可以尝试下能不能用？
> > [image]
> > 驱动下载地址：https://driverdownloads.aquantia.com/
> > [image]
> 
> 请问你现在找到办法在openwrt下驱动aqc111u了吗

没有,至今没有

xiaowonet, 2021-06-03:
同款网卡，现在有驱动吗？