coolsnowwolf/lede #3472: pdnsd配置文件里默认添加opendns，会对DNS请求进行抢答。

fefz, 2020-03-02:
pdns工作的配置文件 /var/etc/pdnsd.conf  内容

```
global{
perm_cache=1024;
cache_dir="/var/pdnsd";
pid_file="/var/run/pdnsd.pid";
run_as="nobody";
server_ip=127.0.0.1;
server_port=5335;
status_ctl=on;
query_method=tcp_only;
min_ttl=1h;
max_ttl=1w;
timeout=10;
neg_domain_pol=on;
proc_limit=2;
procq_limit=8;
par_queries=1;
}
server{
label="ssr-usrdns";
ip=127.0.0.1;
port=5533;
timeout=6;
uptest=none;
interval=10m;
purge_cache=off;
}
server{
label="ssr-pdnsd";
ip=208.67.222.222, 208.67.220.220;
port=5353;
timeout=6;
uptest=none;
interval=10m;
purge_cache=off;
}
```

wireshark抓包结果（涂黑的是自定义的DNS服务器）

[image]

fefz, 2020-03-02:
所以说， #3400 这个issue里提到的sni proxy不工作的问题，可能也是这个原因？

coolsnowwolf, 2020-03-02:
你本地的 127.0.0.1 5533 DNS 服务器的速度居然没有 OpenDNS 回应快？

coolsnowwolf, 2020-03-02:
默认去除了这个 fallback 了。 本意是为了避免有人错误设置 海外 DNS 导致失效的问题

closed 2020-03-02 by coolsnowwolf

tefiszx, 2021-05-24:
> 默认去除了这个 fallback 了。 本意是为了避免有人错误设置 海外 DNS 导致失效的问题

在netflix域名分流列表里面能不能放一个私有dns的设置？