coolsnowwolf/lede #3473: 大佬们，一直编译失败

Atroc-0625, 2020-03-02:
make[3]: Leaving directory '/home/ubuntu/lede/package/kernel/button-hotplug'
time: package/kernel/button-hotplug/compile#0.25#0.18#0.42
make[3]: Entering directory '/home/ubuntu/lede/package/lean/antfs'
mkdir -p /home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ubuntu/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/.pkgdir/kmod-fs-antfs/. /home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-fs-antfs_installed
echo "kmod-fs-antfs" >> /home/ubuntu/lede/staging_dir/target-x86_64_musl/pkginfo/antfs.default.install
make[3]: Leaving directory '/home/ubuntu/lede/package/lean/antfs'
time: package/lean/antfs/compile#0.54#0.54#1.05
make[3]: Entering directory '/home/ubuntu/lede/package/lean/antfs-mount'
mkdir -p /home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ubuntu/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ubuntu/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount/. /home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/'
cp: not writing through dangling symlink '/home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/./sbin/mount.ntfs'
Makefile:28: recipe for target '/home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.antfs-mount_installed' failed
make[3]: *** [/home/ubuntu/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.antfs-mount_installed] Error 1
make[3]: Leaving directory '/home/ubuntu/lede/package/lean/antfs-mount'
time: package/lean/antfs-mount/compile#0.24#0.14#0.36
package/Makefile:111: recipe for target 'package/lean/antfs-mount/compile' failed
make[2]: *** [package/lean/antfs-mount/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:107: recipe for target '/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

KFERMercer, 2020-03-02:
antfs和某些包冲突了

haxc, 2020-03-02:
antfs-mount取消选择这个

iamggs, 2020-03-02:
我编译也是报这个错，而且不知道哪里取消这个antfs-mount，直接在.config里面搜索也搜索不到。

`make[3]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/package/lean/antfs-mount'
mkdir -p /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount/. /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-x86_64_musl/root-x86/'
cp: not writing through dangling symlink '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-x86_64_musl/root-x86/./sbin/mount.ntfs'
make[3]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.antfs-mount_installed] Error 1
Makefile:28: recipe for target '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.antfs-mount_installed' failed
make[2]: *** [package/lean/antfs-mount/compile] Error 2
make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/package/lean/antfs-mount'
make[1]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
time: package/lean/antfs-mount/compile#0.12#0.02#0.15
`

closed 2020-03-02 by coolsnowwolf