coolsnowwolf/lede #3479: ttyd终端会使空闲运行的newifi3平均负载永久保持在0.5-0.6

DongfeiSay, 2020-03-02:
这边排查了好久才找到原来问题居然会是出在小小的ttyd身上，之前包括不限于可能是某驱动导致的，可能是某插件需要fpu只能靠模拟运行导致的，可能是某个内核的版本有问题导致的，在我之前的小米mini x86ttyd都是毫无一丁点影响的 偶然看到koolshare的某贴15L说的把webshell设置从默认的ttyd改为sshellinabox后恢复正常，不然我可能真的找不到原因了

这可让我有丶烦躁

开启ttyd
[image]

关闭ttyd
[image]

DongfeiSay, 2020-03-02:
[image]
有时甚至直接飘到1+

DongfeiSay, 2020-03-02:
话题这里就直接结束了 我也不想说什么了，为什么会这样哦也不知道，想修就修吧，有确实是本应该不该出现的问题出现了 和一些不完善的地方 我这些没技术的小白只能发现 提出一些存在的bug 稍微提供一些信息来希望能让op变得更完善 罢了

DongfeiSay, 2020-03-02:
话题这里就直接结束了我也不想说什么了，为什么会这样哦也不知道，想修就修吧，有确实是本应该不该出现的问题出现了和一些不完善的地方我这些没技术的小白只能发现提出一些存在的bug 稍微提供一些信息来希望能让op变得更完善罢了

closed 2020-03-02 by DongfeiSay

hcym, 2020-03-02:
用这些稀奇古怪的干嘛，默认设置里面我只保留一个科学一个docker

jxph1123, 2020-03-04:
我也是这样，不过我后面直接没编译这个了，没啥用。

ricksuzade-maker, 2020-03-04:
实测发现在x86平台下也存在这个问题，直接屏蔽了算了。