coolsnowwolf/lede #3484: 欣闻大佬引用了anti-AD的去广告项目

gentlyxu, 2020-03-02:
欣闻大佬近期https://github.com/coolsnowwolf/lede/commit/8269399df96759582e9ad707023714e7e2dfbfd7  引用了我们的广告过滤列表——anti-AD，不胜荣幸！

这里一点小小的要求，**希望在相关位置给个我项目的链接**： https://github.com/privacy-protection-tools/anti-AD ，一来方便发现过滤列表有问题的朋友直接反馈到我们这里，进一步提升用户体验；二来我做开源项目，希望有个相互尊重。

hcym, 2020-03-02:
呵呵😄，一直关注那个git，感谢，不过有个小问题，dnsmasq规则应该默认是顶级域名，我现在手机没法给链接，回头去你的地盘发一个老外的GitHub，很规范，就是太多也用不到。

coolsnowwolf, 2020-03-02:
非常感谢，我会在首页注明下的

gentlyxu, 2020-03-02:
> 非常感谢，我会在首页注明下的

感谢！同时这也给了我莫大的动力，我会继续把项目搞的更出色

gentlyxu, 2020-03-02:
> 呵呵😄，一直关注那个git，感谢，不过有个小问题，dnsmasq规则应该默认是顶级域名，我现在手机没法给链接，回头去你的地盘发一个老外的GitHub，很规范，就是太多也用不到。

没太懂你的意思，不过非常想和你交流一下细节欢迎来我的项目提issue

mjyhj, 2020-03-02:
> 欣闻大佬近期[8269399](https://github.com/coolsnowwolf/lede/commit/8269399df96759582e9ad707023714e7e2dfbfd7) 引用了我们的广告过滤列表——anti-AD，不胜荣幸！
> 
> 这里一点小小的要求，**希望在相关位置给个我项目的链接**： https://github.com/privacy-protection-tools/anti-AD ，一来方便发现过滤列表有问题的朋友直接反馈到我们这里，进一步提升用户体验；二来我做开源项目，希望有个相互尊重。

有误杀哦，用了后QQ上的 迷你腾讯新闻会打不开

gentlyxu, 2020-03-02:
https://github.com/privacy-protection-tools/anti-AD 大家来我的项目提issue吧。。。在这里版聊不够尊重别人

closed 2020-03-02 by gentlyxu