coolsnowwolf/lede #3490: SSR  Plus 现在支持 SOCK5 的服务器节点，可以将主路由的运算分担给其机器

coolsnowwolf, 2020-03-02:
搞了个新功能， SSR Plus 现在支持 SOCK5 的服务器节点了，而且也是多线程的

例如你可以把 Trojan 或者 V2RAY 开在内网的一台 Windows 下载机上面，开启允许 SOCK5 局域网代理接入

这样，你的弱鸡主路由器的 SSR Plus 就能连接到这台 SOCKS5 服务上面，运算都在这台 Windows 机器上面了，主路由只负责转发

这样就无需所谓的旁路由了，而且可以利用下载机/NAS 强大的多余性能分担

而且旁路由就算死机，也不会影响你正常上网了~~

[image]

[image]

[image]

luckyyyyy, 2020-03-02:
哈哈 这个没毛病 套娃 分摊

hcym, 2020-03-02:
花了半年才学会hanwckf的单臂，这个要一年了

wujunze, 2020-03-02:
弱鸡主路由建议扔掉  哈哈哈

makepoloSeton, 2020-03-03:
大雕为我们这些穷鬼实在是太费心费力了（大雾）

ddvcx, 2020-03-03:
不服不行

cncaihua, 2020-03-03:
弱机的AR9331表示还能再玩3年，然后还想把垃圾堆里的BCM6358找出来试试。

Myoko, 2020-03-03:
这个功能好！大雕辛苦了！

sancome, 2020-03-03:
很实用的功能，支持！

WellsWang, 2020-03-03:
太好了，这个应该能解决我那个Trojan总是时断时续的问题

haxc, 2020-03-03:
> 弱机的AR9331表示还能再玩3年，然后还想把垃圾堆里的BCM6358找出来试试。

我也有一块 bcm6369吃灰，当年编译的固件lan用不了，要拿出来擦一擦了

BOBINIUNIU, 2020-03-03:
你们搞的这个东西啊，一颗赛艇。但是个人觉得扶墙的木桶应该是线路而不是路由性能吧。
弱鸡路由建议咸鱼出掉然后入手x86主机。

coolsnowwolf, 2020-03-03:
已支持 Socks5 用户名密码

[image]

babylife, 2020-03-03:
不错不错，现在就差均衡负载模块了吧，其他感觉和passwall都一样了

darkrain88, 2020-03-03:
HAproxy 怎么使用最新的源码 , 求带

用几个vps 来聚合trojan的流量

WellsWang, 2020-03-03:
@coolsnowwolf 能否看下这个ISSUE?
https://github.com/coolsnowwolf/lede/issues/3327
Trojan莫名其妙时断时续，当然现在有SOCKS5代理我可以用来workaround，但是我还是想知道原因，是否能给些建议或者告知debug方法
有其他朋友也遇到类似情况。

hcym, 2020-03-03:
遇到，很痛苦😭，好像tcp和udp节点类型不一致的概率大，我是尽量保持一个类型，感觉好一些。

yannanfeigithub, 2020-03-04:
编译的K2P重启动，路由器配置被重置，怎么办？

gfonline, 2020-03-04:
我也遇到了相似的时断时续的问题，在手机和电脑客户端没有出现这个问题

lipwah, 2020-03-04:
为什么重新编译都没有socks5

gfonline, 2020-03-04:
最新版本的的编译选项里ssr plus软件包里并没有socks5的选项啊？

fishbless, 2020-03-04:
这个号，弱鸡主路由有救了

shu0802, 2020-03-05:
SOCKS5这个怎么用？是不是按大雕的做法后，就可以把里面所有的节点都能用上？

mailsex, 2020-03-06:
很实用的功能，支持！

closed 2020-03-06 by coolsnowwolf

cdcchen, 2020-03-30:
这个功能并不实用啊，需要一直开着这台windows电脑。如果有一台nas那是再好不过了。

zhouym233, 2021-01-25:
在menuconfig里找不到sock5相关选项，编译后添加节点时也没有sock5选项，求解