coolsnowwolf/lede #3494: 请教下SSR+新添加的SOCKS玩法只能添加无密码的SOCKS服务器？

tone88, 2020-03-03:
刚刚更新试了下,发现如果SOCKS代理服务器设置了用户名和密码，好像不能使用？   添加节点的时候没有添加用户名和密码的选项。

coolsnowwolf, 2020-03-03:
已添加用户名密码支持

tone88, 2020-03-03:
辛苦大雕了，另外刚刚还发现个bug 切换节点会导致v2ray服务器停止运行。

hcym, 2020-03-03:
大佬，还是慢慢的走，这个版本切换节点问题多，只能重启解决。

coolsnowwolf, 2020-03-03:
那是 v2ray 那边的问题，等那边更新吧

martin0607, 2020-03-03:
终于可以连socks5了，之前一直用kumasocks。。。

martin0607, 2020-03-04:
今天刷了新出的潘多拉，发现Sock5节点连接错误？我是另一台电脑群晖上docker开socks5共享的，
重置了一次发现还是一样？
[image]
[image]
[image]

martin0607, 2020-03-04:
确认了不是群晖也不是V2RAY服务器的问题。
我在内网上的电脑用V2rayn开socks5然后K2P连接，发现日志如下所示：
[image]
看上去是socks5账户密码验证的问题？但是客户端并没有开启账户密码呀。。。
楼主更新源码之后的socks5能正常连接不？

closed 2020-03-06 by coolsnowwolf