coolsnowwolf/lede #3503: nlbwmon负载50%

zwillhill, 2020-03-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue
A：* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：2月十几号的版本编译，nlbwmon负载突然飙升，同时装有wrtbmon（用来配合serverchan），没注意看日志，最近几个版本编译时都没选nlbwmon。
不知是个例还是上述两个包冲突
[image]

### (2) 路由器型号和固件版本

A：K3

### (3) 详细日志

A：

closed 2020-03-06 by coolsnowwolf

hhyygg2009, 2020-03-13:
我的也有同样情况呢,同k3
固件是 OpenWrt R20.2.15 mod by huah0235 v1.5 / LuCI Master (git-20.058.13156-2faf063)
重启服务之后占用正常

zwillhill, 2020-03-13:
> 我的也有同样情况呢,同k3
> 固件是 OpenWrt R20.2.15 mod by huah0235 v1.5 / LuCI Master (git-20.058.13156-2faf063)
> 重启服务之后占用正常

刷机后如果不点带宽统计，能坚持正常几天，然后突然飙升，网上好像有定时重启的脚本，我是觉得没啥用直接卸载了