coolsnowwolf/lede #3506: SSR Plus+ 访问控制，建议用MAC代替LAN IP

swzeng, 2020-03-03:
如题
1. 现在的访问控制是基于 LAN IP的，这样设备一重启，IP变掉了，就无效了，建议用MAC地址替代
2. 在选择局域网设备的时候，现在显示的只有IP，不容易对应到具体设备，建议除了IP外，也显示设备名称
我家现在有下载的盒子(玩客云)，有电视盒子，有NAS(群晖)，还有一堆小米的IoT设备(扫地机啊，灯啊等等)，还是蛮需要LAN的白名单的

dolphinpaopao, 2020-03-03:
可以试试设置ip绑定

swzeng, 2020-03-03:
> 可以试试设置ip绑定

ip绑定跟分配静态ip都可以解决这个问题
不过从用户的角度，设置ssr分流的时候，能在一个界面上完成的，还是方便很多，尤其是现在网络设备越来越多的情况下。passwall就有这个功能，华硕的路由器也在一个界面上显示连接设备的名称，IP，MAC

dolphinpaopao, 2020-03-03:
> > 可以试试设置ip绑定
> 
> ip绑定跟分配静态ip都可以解决这个问题
> 不过从用户的角度，设置ssr分流的时候，能在一个界面上完成的，还是方便很多，尤其是现在网络设备越来越多的情况下。passwall就有这个功能，华硕的路由器也在一个界面上显示连接设备的名称，IP，MAC

平时ssr都开的全局吗？一般来说不设置旁路由网关的话，是否白名单没有影响吧？个人习惯gfw+部分手动黑名单的方式。

swzeng, 2020-03-03:
SSR开的大陆白名单
不过比如家里的一堆小米，下载机，电视盒子，群晖这些都是不需要翻墙的，直接在LAN访问控制里通过MAC白名单过滤掉，不进入SSR Plus+内部的下一步分流(根据destination IP?)，不是更简单？

DongfeiSay, 2020-03-03:
虽说ip绑定跟分配静态ip可以解决这个问题，但是能直接基于MAC来进行控制可以来防止已绑定的mac设备设置为其他ip来绕过访问控制，除非他比较狠把mac也伪装了，还是比较实用的

dolphinpaopao, 2020-03-03:
> SSR开的大陆白名单
> 不过比如家里的一堆小米，下载机，电视盒子，群晖这些都是不需要翻墙的，直接在LAN访问控制里通过MAC白名单过滤掉，不进入SSR Plus+内部的下一步分流(根据destination domain?)，不是更合理？

个人理解，主路由提供DHCP服务，旁路由负责出国的话，不需要出国的设备的网关不指向旁路由就可以了啊。

swzeng, 2020-03-03:
> > SSR开的大陆白名单
> > 不过比如家里的一堆小米，下载机，电视盒子，群晖这些都是不需要翻墙的，直接在LAN访问控制里通过MAC白名单过滤掉，不进入SSR Plus+内部的下一步分流(根据destination domain?)，不是更合理？
> 
> 个人理解，主路由提供DHCP服务，旁路由负责出国的话，不需要出国的设备的网关不指向旁路由就可以了啊。

最开始已经说了，“N1做旁路由，DHCP/DNS/SSR Plus+等等都开在旁路由上”。谁会把家里每一个联网设备都手动设IP，指定网关？肯定是设备自动发现，LAN中由旁路由的DHCP自动分配IP/网关这些啊。

我感觉你把问题复杂化了，从程序功能的角度来说，SSR Plus有分流，只是现在这个分流只基于IP，所以只在接入设备分配静态IP的情况下才起作用。但是如我前面所说，家里的接入设备大部分都是靠旁路由上的DHCP自动分配的，真正不变的只有MAC。所以我的建议是除了基于LAN IP的分流，增加以MAC为基础的分流，增加设备名称的显示。这样用户可以在SSR Plus的访问控制这一个页面上，才能知道哪个设备是哪个，哪个设备需要/不需要走SSR
PS: 可以参照下passwall，就有这个功能，也很好用。

lizhenglin2002, 2020-03-04:
一样有同样的烦恼

mjyhj, 2020-03-05:
自己保留MAC到对应的IP不就好

swzeng, 2020-03-05:
> 自己保留MAC到对应的IP不就好

为什么明明就是应该在这里实现的功能，一定要用户手动的去别的地方调整呢？
既然接到gateway的SSR Plus上来分流，SSR Plus不应该用一个更合理的方式吗？目前基于IP，明显是没有考虑到局域网很多设备是基于DHCP，IP会变这个因素。不应该把这个功能做做好吗？

mjyhj, 2020-03-05:
> > 自己保留MAC到对应的IP不就好
> 
> 为什么明明就是应该在这里实现的功能，一定要用户手动的去别的地方调整呢？
> 既然接到gateway的SSR Plus上来分流，SSR Plus不应该用一个更合理的方式吗？目前基于IP，明显是没有考虑到局域网很多设备是基于DHCP，IP会变这个因素。不应该把这个功能做做好吗？

为什么一定要控制mac，而不是ip。
控制IP是更简单方便。
而且ipset写的是ip

swzeng, 2020-03-05:
> > > 自己保留MAC到对应的IP不就好
> > 
> > 
> > 为什么明明就是应该在这里实现的功能，一定要用户手动的去别的地方调整呢？
> > 既然接到gateway的SSR Plus上来分流，SSR Plus不应该用一个更合理的方式吗？目前基于IP，明显是没有考虑到局域网很多设备是基于DHCP，IP会变这个因素。不应该把这个功能做做好吗？
> 
> 为什么一定要控制mac，而不是ip。
> 控制IP是更简单方便。
> 而且ipset写的是ip

你这是从程序员的角度，而不是从用户的角度。
从用户的角度，局域网DHCP的情况下，设备IP会变，然后时不时的回到SSR下面来修改基于IP的规则吗？

closed 2020-03-06 by coolsnowwolf

swzeng, 2020-03-06:
这个为什么要关？

dolphinpaopao, 2020-03-06:
> 这个为什么要关？

首先感谢作者的无私奉献。
其次为什么作者不能关闭呢？提建议就一定要采纳？心态不对。
最后功能无法实现，过于复杂化，有冲突或者有其他办法可以解决这个问题等等都可以关，这个是作者自由，你并不能要求作者做什么。这不是个人定制化的项目。满足多数的需求为前提。

swzeng, 2020-03-06:
我也是做过专业的嵌入式软件测试的，这功能明明就是没做好，我还挺认真前前后后弄明白问题出在哪，认认真真给了建议，一句话不说直接关掉，可以的，很大神
至于你，请你回帖先看看明白别人到底说了什么，不要一跳出来就指责这个那个，先看懂，在说话，ok？
言尽于此

DongfeiSay, 2020-03-06:
> 我也是做过专业的嵌入式软件测试的，这功能明明就是没做好，我还挺认真前前后后弄明白问题出在哪，认认真真给了建议，一句话不说直接关掉，可以的，很大神
> 至于你，请你回帖先看看明白别人到底说了什么，不要一跳出来就指责这个那个，先看懂，在说话，ok？
> 言尽于此

刚才还一百多个突然全部关掉了😂

dolphinpaopao, 2020-03-06:
> 我也是做过专业的嵌入式软件测试的，这功能明明就是没做好，我还挺认真前前后后弄明白问题出在哪，认认真真给了建议，一句话不说直接关掉，可以的，很大神
> 至于你，请你回帖先看看明白别人到底说了什么，不要一跳出来就指责这个那个，先看懂，在说话，ok？
> 言尽于此

很抱歉不小心激怒了您，我为前面说的话表示歉意。既然您是专业人士也有好的想法，这是一个来源项目，您完全可以通过提交PR的形式更有利于本项目的发展。万分感谢。

xiaowu01, 2020-08-12:
可以的话，IP最好也可以以IP段的形式表述，如：x.x.x.50-100,或x.x.x.50/100，不然50台，就得加50个IP。

l123456789jy, 2022-01-11:
OpenWrt 使用 Dnsmasq 来提供 DHCP 服务，根据客户端 MAC 地址的哈希值来为客户端分配 IP 地址，只要不选择顺序分配 IP 地址，就不会是随机的，所以用IP也是没问题的@swzeng

allen2424, 2022-06-18:
> 首先感谢作者！这是个特别好用的插件，强大稳定，是不可多得的精品 我的情况是，N1做旁路由，DHCP/DNS/SSR Plus+等等都开在旁路由上，下面通过华硕的路由器挂了下载的盒子(玩客云)，电视盒子，NAS(群晖)，还有一堆小米的IoT设备(扫地机啊，灯啊等等)，都是需要LAN的白名单的，所以有两个小小的建议：
> 
> 1. 现在LAN的访问控制是基于IP的，这样设备一重启，IP变掉了，就无效了，建议用MAC地址替代
> 2. 在选择局域网设备的时候，现在显示的只有IP，不容易对应到具体设备，建议除了IP外，也显示设备名称

你应该先设置IP和MAC地址绑定，之后就没这些问题了，这是网络的基本操作。