coolsnowwolf/lede #3509: 连接外接显示器有信号输出但是黑屏

WCJ23333, 2020-03-03:
### (1) 具体问题

A：x86 3月3日源码，intel核显 连接外接显示器有信号输出但是黑屏，怀疑是输出分辨率超了，但我没有第二个显示器验证

### (2) 路由器型号和固件版本

A：

主机名 | OpenWrt
-- | --
主机型号 | Intel(R) Core(TM) i5-3470S CPU @ 2.90GHz : 4 Core 4 Thread (CpuMark : 82413.337225 Scores)
CPU信息 | 3078.130 MHz +38.0°C
固件版本 | OpenWrt R20.2.15 / LuCI Master (git-20.062.50551-a6189c5)
内核版本 | 4.19.106
本地时间 | Tue Mar 3 16:01:44 2020

unclehu87, 2020-03-03:
主机型号 | Intel(R) Celeron(R) CPU J1900 @ 1.99GHz : 4 Core 4 Thread
-- | --
CPU信息 | 1999.294 MHz +40.0°C
固件版本 | OpenWrt R20.2.15 / LuCI Master (git-20.062.50551-a6189c5)
内核版本 | 4.19.106

我和你的问题一样。连着电视看到系统启动完成后就黑了

3mile, 2020-03-04:
笔记本做路由，屏幕也没显示。ssh正常

AmadeusGhost, 2020-03-04:
Refers to 1c6e4da87a1e20c0d810dc8dcbdf373764edd3d9
估计这个commit有问题

closed 2020-03-06 by coolsnowwolf

anythink-wx, 2020-03-25:
我也遇到了-。- 能从后台升级解决吗？

yjssiqpl333, 2020-06-12:
> 大神修复完了，刚从新编译一下没有这个问题了

我也是3865u启动显示器黑屏?请问怎么处理