coolsnowwolf/lede #3515: dockerman连不上外网，内存只能最大2G

bin20088, 2020-03-03:
之前一直用的docker-ce留下的opt分区，平稳过渡到dockerman没问题，今天整个把软路由格式化重新安装X86，发现docker-web没了，只有dockerman，才发现有这两个问题：
1、dockerman连不上外网
搜索了issue，确实存在类似的issue：
https://github.com/lisaac/luci-app-dockerman/issues/6

2、内存最大只能使用2G，我的软路由有6G。
搜索了issue，确实存在类似的issue：
https://github.com/lisaac/luci-app-dockerman/issues/16
这个虽然问的不一样，但是差不多，后面要是内存填了超过2G，无法保存。

KFERMercer, 2020-03-03:
rm -rf /var/lib/docker && service dockerd restart

lloyd18, 2020-03-03:
下面的iss我发的，已经pr给大雕修复，看他接不接受了

KFERMercer, 2020-03-03:
👍

bin20088, 2020-03-03:
> rm -rf / var / lib / docker &&服务dockerd重新启动

> rm -rf /var/lib/docker && service dockerd restart

没有此路径  /var/lib/docker

yuban10703, 2020-03-03:
https://github.com/lisaac/luci-app-dockerman/issues/15
内存显示好像是依赖的问题

yuban10703, 2020-03-03:
> rm -rf /var/lib/docker && service dockerd restart

我从上一个版本升级的固件可以用这个解决
我重新刷的固件执行了貌似还是不行............

yuban10703, 2020-03-03:
> > rm -rf /var/lib/docker && service dockerd restart
> 
> 我从上一个版本升级的固件可以用这个解决
> 我重新刷的固件执行了貌似还是不行............

允许wan口访问后又执行一次可以了........

bin20088, 2020-03-03:
> 之前一直用的docker-ce留下的opt分区，平稳过渡到dockerman没问题，今天整个把软路由格式化重新安装X86，发现docker-web没了，只有dockerman，才发现有这两个问题：
> 1、dockerman连不上外网
> 搜索了issue，确实存在类似的issue：
> [lisaac/luci-app-dockerman#6](https://github.com/lisaac/luci-app-dockerman/issues/6)
> 
> 2、内存最大只能使用2G，我的软路由有6G。
> 搜索了issue，确实存在类似的issue：
> [lisaac/luci-app-dockerman#16](https://github.com/lisaac/luci-app-dockerman/issues/16)
> 这个虽然问的不一样，但是差不多，后面要是内存填了超过2G，无法保存。

不好意思，误导你们了，能连外网，是我搞错了。
现在就第二个问题，希望我的6G内存能全部发挥起来。现在只能最大设置2G

yuban10703, 2020-03-03:
> 不好意思，误导你们了，能连外网，是我搞错了。
> 现在就第二个问题，希望我的6G内存能全部发挥起来。现在只能最大设置2G

https://github.com/lisaac/luci-app-dockerman/issues/15
好像是这个问题

jy3344511, 2020-03-04:
[
{
"Id": "d29da1632470f569b7c8729e644192936345aa77583cd6917be880e69bec84b1",
"Created": "2020-03-04T08:00:18.045999301Z",
"Path": "/init",
"Args": [],
"State": {
"Status": "running",
"Running": true,
"Paused": false,
"Restarting": false,
"OOMKilled": false,
"Dead": false,
"Pid": 3909,
"ExitCode": 0,
"Error": "",
"StartedAt": "2020-03-04T08:01:05.906203953Z",
"FinishedAt": "0001-01-01T00:00:00Z"
},
"Image": "sha256:835f7918dfb8798571fdd7b9be3f97b204f9af1ea7fd06074af4ebfae1bf259e",
"ResolvConfPath": "/opt/docker/containers/d29da1632470f569b7c8729e644192936345aa77583cd6917be880e69bec84b1/resolv.conf",
"HostnamePath": "/opt/docker/containers/d29da1632470f569b7c8729e644192936345aa77583cd6917be880e69bec84b1/hostname",
"HostsPath": "/opt/docker/containers/d29da1632470f569b7c8729e644192936345aa77583cd6917be880e69bec84b1/hosts",
"LogPath": "/opt/docker/containers/d29da1632470f569b7c8729e644192936345aa77583cd6917be880e69bec84b1/d29da1632470f569b7c8729e644192936345aa77583cd6917be880e69bec84b1-json.log",
"Name": "/qbittorrent",
"RestartCount": 0,
"Driver": "overlay2",
"Platform": "linux",
"MountLabel": "",
"ProcessLabel": "",
"AppArmorProfile": "",
"ExecIDs": null,
"HostConfig": {
"Binds": [
"/opt/docker:/config",
"/mnt/sda1:/downloads"
],
"ContainerIDFile": "",
"LogConfig": {
"Type": "json-file",
"Config": {}
},
"NetworkMode": "default",
"PortBindings": {
"6881/tcp": [
{
"HostIp": "",
"HostPort": "6881"
}
],
"6881/udp": [
{
"HostIp": "",
"HostPort": "6881"
}
],
"8989/tcp": [
{
"HostIp": "",
"HostPort": "8989"
}
]
},
"RestartPolicy": {
"Name": "unless-stopped",
"MaximumRetryCount": 0
},
"AutoRemove": false,
"VolumeDriver": "",
"VolumesFrom": null,
[... 188 lines cut ...]
"Networks": {
"bridge": {
"IPAMConfig": null,
"Links": null,
"Aliases": null,
"NetworkID": "e1727e748be52d67bf3bbf9394e450dc5fea1cea7ddc8d37cc855d7500c0f5f8",
"EndpointID": "476010501c178aa9387da12c023abaf66c6e8f9cd072750ae252d0d7910fe51c",
"Gateway": "172.17.0.1",
"IPAddress": "172.17.0.2",
"IPPrefixLen": 16,
"IPv6Gateway": "",
"GlobalIPv6Address": "",
"GlobalIPv6PrefixLen": 0,
"MacAddress": "02:42:ac:11:00:02",
"DriverOpts": null
}
}
}
}
]

之前用的docker，修改为host后，使用qbittorrent可以下载，有速度。后来升级到dockerman后，默认使用的是bridge，直接就可以下载BT。在2020.03.03日更新后，dockerman上运行的容器qbittorrent，就无法下载了。用过iptables -t nat -A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE、ip link del docker0
reboot，都没有用。

另外在创建容器时，添加--net=host，提示无法创建，在dockerman中怎么修改成host？

closed 2020-03-06 by coolsnowwolf