coolsnowwolf/lede #3517: 关于Lean特色插件做成源的可能性

yfdoor, 2020-03-03:
首先很感谢Lean的这个项目，我也是在这个项目才学习的开始自己编译固件。
然而现在经常在 issue 和 PR 区看到很多人其实就是为了 SSR 或是某个插件才使用的整个包。而也正是这些插件出错的概率很大，也许是由于每个人编译的过程导致。
但是不知道大佬们是否想过，我们为什么不做成一个 package 的源呢？将我们的这些特色插件放到插件源中，然后大家定义 opkg/customfeeds.conf， 来加载插件源。直接来安装插件即可？
当然可能把问题想的简单了。。仅提个小建议供参考。

Aes64X, 2020-03-03:
特色警告！

hxlls, 2020-03-03:
整个固件能使得兼容性和出错率更小，更容易使用，做成源，明天来一个XXX不能用,xxxx 固件上，关键是这个XXX固件的源码还不是官方固件，谁有这精力去维护这么多？想要自己把对应源码丢过去自己编译不就完事了？单独编译IPK 又不是不能编译

Her0R0cket, 2020-03-04:
现在luci大改版,还有一些依赖包版本不兼容,所以目前最好不要,如果哪天这边luci也升级了的话,还可以试试

iyuangang, 2020-03-04:
如果是 OpenWrt 19.07.1 稳定版的话，我的项目 [OpenWrt Stable + Lean's package](https://github.com/iyuangang/openwrt) 有编译完成的 `ssr plus + ` 和 `adbybhyplus` 完整包及依赖包供下载。
在`actions` 里找最新的`artifacts`就OK。 目前提供 `x86_64 `和 `树莓派3b/3b+`  版本。
当然也可以fork后修改 `workflows` 适配你自己的硬件。

tuTuStream, 2020-03-04:
> 
> 
> 如果是 OpenWrt 19.07.1 稳定版的话，我的项目 [OpenWrt Stable + Lean's package](https://github.com/iyuangang/openwrt) 有编译完成的 `ssr plus + ` 和 `adbybyplus` 完整包及依赖包供下载。
> 在`actions` 里找最新的`artifacts`就OK。 目前提供 `x86_64 `和 `树莓派3b/3b+` 版本。
> 当然也可以fork后修改 `workflows` 适配你自己的硬件或编译更多L大项目内的特色软件。

这个包能不能把所有这边的软件都打包啊？这样不就可以做成源了吗？

iyuangang, 2020-03-04:
> > 如果是 OpenWrt 19.07.1 稳定版的话，我的项目 [OpenWrt Stable + Lean's package](https://github.com/iyuangang/openwrt) 有编译完成的 `ssr plus + ` 和 `adbybyplus` 完整包及依赖包供下载。
> > 在`actions` 里找最新的`artifacts`就OK。 目前提供 `x86_64 `和 `树莓派3b/3b+` 版本。
> > 当然也可以fork后修改 `workflows` 适配你自己的硬件或编译更多L大项目内的特色软件。
> 
> 这个包能不能把所有这边的软件都打包啊？这样不就可以做成源了吗？

1. 弄个源有喝茶风险。
2. 按自己需求编译固件，成功刷入后实现自己预期功能的快乐，比谈恋爱还爽。

yfdoor, 2020-03-04:
> 1. 弄个源有喝茶风险。
> 2. 按自己需求编译固件，成功刷入后实现自己预期功能的快乐，比谈恋爱还爽。

还有被喝茶的风险啊，这个还真不了解。嗯，自己编译确实有很多乐趣，也是最近在添加其他源的时候，突然想到这个问题。要是有风险还是别搞了。。

hubanya, 2020-03-05:
为什么不能做成ipk，像官方提交？

hyird, 2020-03-05:
https://github.com/hyird/action-openwrt.git  我的项目每5个小时自动编译 自动push到 分支
目前有x64 ipq40xx mt7621 全部插件
是在 分支里面哟

txdywy, 2020-03-05:
赞!

txdywy, 2020-03-05:
赞!

closed 2020-03-06 by coolsnowwolf