coolsnowwolf/lede #3519: UEFI Support?

crabhit, 2020-03-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：menuconfig 里没有支持UEFI的选项

### (2) 路由器型号和固件版本

A：Hyper-V Gen2. VHD

### (3) 详细日志

A：无。

hue715, 2020-03-03:
这个就是
[image]

crabhit, 2020-03-04:
> 这个就是
> [image]

哦 感谢。那生成VHD的时候能支持UEFI吗? 好像生成的VHD在hyperv里无法启动 说是不支持UEFI

norikohxb, 2020-03-04:
请问楼上2个大神这个回帖怎么发布带图片的。我怎么找不到上传按钮？

hue715, 2020-03-04:
> > 这个就是
> > [image]
> 
> 哦 感谢。那生成VHD的时候能支持UEFI吗? 好像生成的VHD在hyperv里无法启动 说是不支持UEFI

图片中的选项就是增加EFI引导支持的（只针对Linux x86和x64），我对hyperv不了解，没法解答不能启动的问题。

hue715, 2020-03-04:
> 请问楼上2个大神这个回帖怎么发布带图片的。我怎么找不到上传按钮？

我也是初学者😂，其实发布图片的方法网上搜索一下也可以知道，只需要把图片拖入，就会自动上传图片并添加。

closed 2020-03-05 by coolsnowwolf