coolsnowwolf/lede #3520: 关于固件稳定性问题的请教

Guxing0321, 2020-03-03:
Hi Team,

最近一段时间，自己编译固件，发现每一两天路由器都会重启一次
路由器：newifi3 d2 和 极路由b70

目前我在考虑有可能出问题的地方：
 
-  **编译时候软件包源的问题**
    编译时候使用了代理，但是，不确定所有的包都下载下来了，能确定的是最终固件是编译出来，没 有报明显错误，如果国内有源的话，帮我提供一个，谢谢

-  **配置的错误，软件包冲突什么的，我拷贝了一份config.seed文件，帮我看下有没有问题**
    [config.seed.txt](https://github.com/coolsnowwolf/lede/files/4281479/config.seed.txt)

-  **硬件问题**
   我也考虑过是硬件的问题，但是，我再恩山论坛里面找了一个newifi3 d2 的固件，用过，确实没有出现经常性莫名的重启，说明硬件是没问题的

其他方面我暂时还没考虑到，希望能帮我查出来到底是因为什么问题

系统的日志我贴在贴在附件中
[内核log.txt](https://github.com/coolsnowwolf/lede/files/4281581/log.txt)
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/4281582/default.txt)

BR
Robin

coolsnowwolf, 2020-03-03:
不多拨就行了

Guxing0321, 2020-03-03:
Hi 

好的，我把多拨关掉试试，谢谢

BR
Robin

closed 2020-03-04 by coolsnowwolf