coolsnowwolf/lede #3522: 【建议】luci-app-ssr-plus插件中的默认广告地址改为dnsmasq格式

gentlyxu, 2020-03-03:
我是[anti-ad广告过滤列表项目](https://github.com/privacy-protection-tools/anti-AD)的作者，看到贵项目引用了过滤列表，荣幸之余，看了点ssr-plus的代码，似乎有依赖dnsmasq，因此提一点点建议

当前默认引用的[easylist.txt](https://gitee.com/privacy-protection-tools/anti-ad/raw/master/easylist.txt) ，这个列表实际上是adblock风格的，有正则表达式和通配符支持，dnsmasq并不支持这些。另有列表 [anti-ad-for-dnsmasq.conf](https://gitee.com/privacy-protection-tools/anti-ad/raw/master/anti-ad-for-dnsmasq.conf) ，他们都是由相同的上游列表生成的，过滤效果几乎一样，又因后者是专门的dnsmasq格式，兼容性更好。

故建议引用 https://gitee.com/privacy-protection-tools/anti-ad/raw/master/anti-ad-for-dnsmasq.conf

coolsnowwolf, 2020-03-03:
因为大部分的 订阅其实都是 adblock 风格的，所以一开始就做成这样的

如果为了这个改为 dnsmasq格式 ，那实际上只有你这个订阅源可用了

PS: 不知道你那边能不能单独做个 anti-ad-for-ssr-plus.txt  呢。。。因为还是想保持兼容性

zgs360, 2020-03-04:
启用广告屏蔽后，一些网页股票时线图就看不到了，如网易的股票时线图网页，就个过滤文件不是太有好！

coolsnowwolf, 2020-03-04:
根据建议，新加了对 DNSMASQ  格式的去广告支持，自动判断

https://gitee.com/privacy-protection-tools/anti-ad/raw/master/anti-ad-for-dnsmasq.conf

gpeditmsc1986, 2020-03-04:
这么快就加上了，这速度绝了！！

gentlyxu, 2020-03-04:
感谢！

closed 2020-03-04 by coolsnowwolf