coolsnowwolf/lede #3523: 官方OpenWrt 18.06.8使用广告屏蔽大师 Plus+不能更新规则

S0MEW0LF, 2020-03-03:
使用的是整个lede编译后在lede/bin/packages/x86_64/base上面找到的adbyby_2.7-20181023_x86_64.ipk，luci-app-adbyby-plus_2.0-45_all.ipk，luci-i18n-adbyby-plus-zh-cn_2.0-45_all.ipk
我是小白，也不太懂这个，尝试了一下SSH里面运行升级命令，结果：
root@OpenWrt:~# /usr/share/adbyby/adupdate.sh
/usr/share/adbyby/adupdate.sh: line 6: wget-ssl: not found
MD5 not match! Need update!
/usr/share/adbyby/adupdate.sh: line 28: wget-ssl: not found
/usr/share/adbyby/adupdate.sh: line 28: wget-ssl: not found

tuTuStream, 2020-03-04:
`wget-ssl: not found` 很明显啊，安装wget 就好了。
[image]

SharerMax, 2020-03-04:
少依赖 找一下编译后生成的 wget 的ipk，也安装上

closed 2020-03-04 by coolsnowwolf