coolsnowwolf/lede #3526: xiaomi r3g的编译总是失败，求大佬帮忙看看如何解决

2287405159, 2020-03-04:
小米r3g的编译总是报错
OSError: [Errno 8] Exec format error
tools/v8_gypfiles/generate_bytecode_builtins_list.target.mk:13: recipe for target '/home/wh/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h' failed
make[5]: *** [/home/wh/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h] Error 1
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/wh/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/wh/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built' failed
make[3]: *** [/home/wh/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/wh/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#8.39#12.34#21.12
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/wh/lede'
package/Makefile:107: recipe for target '/home/wh/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/wh/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wh/lede'
/home/wh/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

jxph1123, 2020-03-04:
我就是r3g，没问题的，首次编译要翻墙才行。

2287405159, 2020-03-04:
> 我就是r3g，没问题的，首次编译要翻墙才行。

我是先全局用make download然后再用的make V=s

Aes64X, 2020-03-04:
./scripts/feeds update -a 
./scripts/feeds install -a
你经常看到的只是源码的一部分更新，好好看看别的。

threekidney, 2020-03-04:
请问楼上的，你们编译的R3G会只有一个5G 无线吗？我这边编译出来刷完后就只有一个5G无线

closed 2020-03-06 by coolsnowwolf