coolsnowwolf/lede #3527: 建议还是把720nv3加回来支持吧

Mcxunyun666, 2020-03-04:
看到tiny-tp-link.mk有720nv3了，但似乎编译时还是找不到，config-4.9里写的是 TP-Link wr720n v3 is not set.

Mcxunyun666, 2020-03-04:
是config-4.9.mk有wr720nv3但被注释掉了。写的是TP-LINK_WR720NV3 is not set.
config-4.14中也没有。
说明下：我希望能加回支持的原因是希望让这些老设备能得到新生。毕竟总有些DIY玩家会把他们爆改的。我改了32m rom + 128 mb ram 但找不到新固件去刷，自己编译也没有找到支持的源码。所以才会找l大来帮忙。希望l大能重新加回支持，安慰下我们这些爱好者。幸苦了。

haxc, 2020-03-04:
> 是config-4.9.mk有wr720nv3但被注释掉了。写的是TP-LINK_WR720NV3 is not set.
> config-4.14中也没有。
> 说明下：我希望能加回支持的原因是希望让这些老设备能得到新生。毕竟总有些DIY玩家会把他们爆改的。我改了32m rom + 128 mb ram 但找不到新固件去刷，自己编译也没有找到支持的源码。所以才会找l大来帮忙。希望l大能重新加回支持，安慰下我们这些爱好者。幸苦了。

define Device/tl-wr720n-v3-16m
  $(Device/tplink-16mlzma)
  DEVICE_TITLE := TP-LINK TL-WR720N v3 16M
  DEVICE_PACKAGES := kmod-usb-core kmod-usb2
  BOARDNAME := TL-WR720N-v3
  DEVICE_PROFILE := TLWR720
  TPLINK_HWID := 0x07200103
  CONSOLE := ttyATH0,115200
endef
TARGET_DEVICES += tl-wr720n-v3-16m

720n v3有的啊！在tiny-tp-link，自己找找

FishSlide, 2020-03-06:
有720n v3，用 /720n 查找，第一个出来的就是720n v3，v4还是在v3的基础上改的呢。我昨天刚编译完了。

closed 2020-03-06 by coolsnowwolf