coolsnowwolf/lede #3529: ssr里面的广告拦截更新一波以后，白名单模式无法看世界，GFW模式可以

kl9800, 2020-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ssr里面的广告拦截更新一波以后，“绕过”模式无法看世界，GFW模式可以

### (2) 路由器型号和固件版本

A：X86  J1900    OpenWrt R20.2.15 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志

A：
Wed Mar  4 11:20:17 2020 daemon.err dnscrypt-proxy[3941]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar  4 11:21:35 2020 daemon.info dnscrypt-proxy[3941]: dnscrypt-proxy Refetching server certificates
Wed Mar  4 11:21:51 2020 daemon.err dnscrypt-proxy[3941]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar  4 11:23:12 2020 daemon.info dnscrypt-proxy[3941]: dnscrypt-proxy Refetching server certificates
Wed Mar  4 11:23:17 2020 daemon.warn pdnsd[25986]: Caught signal 15. Exiting.
Wed Mar  4 11:23:17 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar  4 11:23:17 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar  4 11:23:20 2020 cron.info crond[31981]: crond (busybox 1.30.0) started, log level 8
Wed Mar  4 11:23:21 2020 user.err ssr-rules[32227]: Start failed!
Wed Mar  4 11:23:23 2020 daemon.info pdnsd[32727]: pdnsd-1.2.9b-par starting.
Wed Mar  4 11:23:23 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar  4 11:23:23 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar  4 11:23:27 2020 daemon.err dnscrypt-proxy[3941]: dnscrypt-proxy Unable to retrieve server certificates

TheStoneOasis, 2020-03-04:
原来是这两个在冲突，我说怎么绕过模式连不上google

coolsnowwolf, 2020-03-04:
魔改固件自己找作者

kl9800, 2020-03-04:
我不知道你说的魔改是哪个插件，我是今天上午最新编译的新固件，已经去掉了DNSforward等多余插件，完完全全标准编译，标准ssr-plus，还是会这样，你是说广告过滤这个模块要找作者吗？

coolsnowwolf, 2020-03-04:
dnscrypt-proxy Refetching server certificates

coolsnowwolf, 2020-03-04:
明显是 dnscrypt-proxy Unable to retrieve server certificates 这一步有问题了

kl9800, 2020-03-04:
但是我没有安装这个插件，邪门了，我已经去掉了。。并且重新刷的系统。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。

kl9800, 2020-03-04:
我再仔细查查吧，谢谢大佬

kl9800, 2020-03-04:
我滴妈耶！！！！我少去了一个。。。

EstFoxs, 2020-03-04:
china_ip_list 问题，更新之后只能gfw，白名单不行

EstFoxs, 2020-03-04:
https://github.com/17mon/china_ip_list/raw/master/china_ip_list.txt

kl9800, 2020-03-04:
谢谢

kl9800, 2020-03-04:
> https://github.com/17mon/china_ip_list/raw/master/china_ip_list.txt

那有没有更靠谱的源？

SharerMax, 2020-03-04:
https://github.com/pexcn/daily

kl9800, 2020-03-04:
> https://github.com/pexcn/daily

谢谢

SharerMax, 2020-03-04:
https://ispip.clang.cn/
 来自：官方TG群

kl9800, 2020-03-04:
> https://ispip.clang.cn/
> 来自：官方TG群

66666666666666666666

coolsnowwolf, 2020-03-04:
https://ispip.clang.cn/all_cn.txt

kl9800, 2020-03-04:
> https://ispip.clang.cn/all_cn.txt

谢谢各位大佬！！！！

kl9800, 2020-03-04:
果然是IP列表问题，换了个源，好了，谢谢各位！！

closed 2020-03-04 by kl9800