coolsnowwolf/lede #3533: 最新版159-1的ssr需要全局才能科学上网

R3pl4c3r, 2020-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新版159-1的ssr需要全局才能科学上网

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：Wed Mar  4 12:18:41 2020 daemon.err pdnsd[7786]: Could not bind tcp socket: Address in use
Wed Mar  4 12:18:41 2020 daemon.err pdnsd[7786]: Could not bind to udp socket: Address in use
Wed Mar  4 12:18:41 2020 daemon.err pdnsd[7786]: tcp and udp initialization failed. Exiting.
Wed Mar  4 12:18:41 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar  4 12:18:41 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar  4 12:18:41 2020 kern.info /usr/bin/ssr-server[7926]: UDP relay enabled
Wed Mar  4 12:18:41 2020 kern.info /usr/bin/ssr-server[7926]: initializing ciphers... rc4-md5
Wed Mar  4 12:18:41 2020 kern.info /usr/bin/ssr-server[7926]: tcp port reuse enabled
Wed Mar  4 12:18:41 2020 kern.info /usr/bin/ssr-server[7926]: udp port reuse enabled
Wed Mar  4 12:18:41 2020 kern.info /usr/bin/ssr-server[7926]: listening at 0.0.0.0:65500
Wed Mar  4 12:18:41 2020 kern.info /usr/bin/ssr-server[7926]: running from root user
Wed Mar  4 12:18:41 2020 daemon.err uhttpd[3479]: Collected errors:
Wed Mar  4 12:18:41 2020 daemon.err uhttpd[3479]:  * resolve_conffiles: Existing conffile /etc/china_ssr.txt is different from the conffile in the new package. The new conffile will be placed at /etc/china_ssr.txt-opkg.
Wed Mar  4 12:18:41 2020 daemon.err uhttpd[3479]:  * resolve_conffiles: Existing conffile /etc/config/shadowsocksr is different from the conffile in the new package. The new conffile will be placed at /etc/config/shadowsocksr-opkg.
Wed Mar  4 12:19:17 2020 user.warn ddns-scripts[31636]: myddns_ipv4: Updating IP at DDNS provider failed - starting retry 83/0
Wed Mar  4 12:21:21 2020 authpriv.info dropbear[6915]: Exit (root): Exited normally
Wed Mar  4 12:26:21 2020 kern.err /usr/bin/ssr-redir[7762]: server recv: Connection reset by peer
Wed Mar  4 12:26:21 2020 kern.err /usr/bin/ssr-redir[7752]: server recv: Connection reset by peer
Wed Mar  4 12:29:29 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar  4 12:29:29 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar  4 12:29:29 2020 cron.info crond[15965]: crond (busybox 1.30.0) started, log level 5
Wed Mar  4 12:29:29 2020 user.err ssr-rules[16013]: Start failed!

SharerMax, 2020-03-04:
不保留配置升级试试

R3pl4c3r, 2020-03-04:
> 不保留配置升级试试

看样子光升级app不行

R3pl4c3r, 2020-03-04:
问题解决了，是国内IP的问题，把自定义国内IP的那个勾取掉再更新就OK

R3pl4c3r, 2020-03-04:
md玄学问题

coolsnowwolf, 2020-03-04:
国内自定义的IP规则库崩了，换成下面这个官方地址

[image]

https://ispip.clang.cn/all_cn.txt

R3pl4c3r, 2020-03-04:
> 国内自定义的IP规则库崩了，换成下面这个官方地址
> 
> [image]
> 
> https://ispip.clang.cn/all_cn.txt

感谢

closed 2020-03-04 by R3pl4c3r