coolsnowwolf/lede #3536: lean大什么时候能够在ssrplus的去广告里面屏蔽youtube的而广告

norikohxb, 2020-03-04:
lean大什么时候能够在ssrplus的去广告里面屏蔽youtube的而广告
我说的不是网页的youtube的广告那个有插件可以去除。
我说的是youtube app播放youtube视频中间的广告，
不知道这个能否实现
我知道虽然只有几秒，但是有时候还要伸手出来按跳过广告冬天被窝挺冷的。哈哈

R3pl4c3r, 2020-03-04:
请各位用户阅读以下内容再决定是否发起pull request
[关于拉取请求 - GitHub Help](https://help.github.com/cn/github/collaborating-with-issues-and-pull-requests/about-pull-requests)

>拉取请求可让您在 GitHub 上向他人告知**您已经推送到仓库中分支的更改**。 在拉取请求打开后，您可以与协作者讨论并审查潜在更改，在更改合并到基本分支之前添加跟进**提交**。
你当前的操作会 **向** *Lean的代码* 中 **提交** *你的代码* 。如果你想同步Lean的代码到你的仓库请看以下内容

[同步fork的项目的代码 - 教程](https://blog.csdn.net/qq1332479771/article/details/56087333)

R3pl4c3r, 2020-03-04:
> lean大什么时候能够在ssrplus的去广告里面屏蔽youtube的而广告
> 我说的不是网页的youtube的广告那个有插件可以去除。
> 我说的是youtube app播放youtube视频中间的广告，
> 不知道这个能否实现
> 我知道虽然只有几秒，但是有时候还要伸手出来按跳过广告冬天被窝挺冷的。哈哈

sorry刚才回复错了

norikohxb, 2020-03-04:
chrome下面的youtube可以通过插件过滤掉了，但是平时手机或者是平板看，有些视频广告特别多。有些视频比较少，能否在ssrplus里面对于youtube的视频中间的那些广告做去除？谢谢了

edmondfr, 2020-03-04:
手机如果是安卓的，可以安装youtube advanced app完全去广告

garypang13, 2020-03-04:
> 手机如果是安卓的，可以安装youtube advanced app完全去广告

前提是装了magisk,这就难倒一大半了.

Wilson-coder, 2020-03-04:
Youtube Vanced 非Root版一样可以滤广告。应用为啥名字叫vanced，而不是advanced，应该就是取没有“ad”的意思。

norikohxb, 2020-03-04:
咱们lean大在路由器上面直接集成搞定可能性很小吗？

SoPudge, 2020-03-04:
给人家留点挣钱的余地吧……油管广告比国内的强多了，至少不会120秒，几秒钟的事

hcym, 2020-03-04:
手机上面是烦，用go-lite版本好一些，盒子用smart YouTube不错

norikohxb, 2020-03-04:
> 给人家留点挣钱的余地吧……油管广告比国内的强多了，至少不会120秒，几秒钟的事

很高觉悟的都是去交youtube会员的。我也都是点击右下方的那个跳过，说实话发布的人也没有赚到广告费。你有这个觉悟可以去缴费嘛

closed 2020-03-04 by coolsnowwolf

shijiezhou1, 2022-12-21:
> 

这个软件是否安全？