coolsnowwolf/lede #3541: 使用P3TERX大老在线编译问题

nb12nb34, 2020-03-04:
原来使用P3TERX大老在线编译没什么问题，可是最近更新106内核后，同样编译就不能使用，使用x86编译，问题是虚拟下能够使用，但是强制关机后就不能用了。。实体机一旦开机然后就没有启动分区了。。求大佬帮忙解决。。。本人小白，本着学习的精神，不想做伸手党。。。谢谢了啦。。

EnnawYang, 2020-03-04:
[image]

P3TERX, 2020-03-04:
[image]

nb12nb34, 2020-03-04:
是的。。。不知道问题出在哪？还是我的设置问题。。可是同样设置以前就能用。。。现在就over了。。。。

necoic, 2020-03-04:
[image]

R3pl4c3r, 2020-03-04:
> 原来使用P3TERX大老在线编译没什么问题，可是最近更新106内核后，同样编译就不能使用，使用x86编译，问题是虚拟下能够使用，但是强制关机后就不能用了。。实体机一旦开机然后就没有启动分区了。。求大佬帮忙解决。。。本人小白，本着学习的精神，不想做伸手党。。。谢谢了啦。。

你用的是rootfs还是combined，rootfs是没有引导的，如果你用的是rootfs，建议换成combined

nb12nb34, 2020-03-04:
我用openwrt-x86-64-uefi-gpt-squashfs.img.gz引导。。。下午有重新编译了一次。。。暂时又可以使用。。这次把openvpn-sever 编译上了。。目前不知道是不是这个原因。。。晚上在跑一次。。重新测试下。。。。谢谢大家的回答。。。

closed 2020-03-06 by coolsnowwolf